JMRI/JMRI #3494: ConcurrentModificationException on startup

silverailscolo, 2017-05-12:
On a first build of current master 4.7.4ish (in IntelliJ 2017.1 IDE on macOS with Loconet Sim and MERG Sim, English locale) I frequently see this during PanelPro startup, stopping with the splash screen remaining on screen plus the LocoNet Simulator pane and only the PanelPro menu item (no main PanelPro):

```
2017-05-12 18:36:34,139 node.NodeIdentity                     INFO  - Using jmri-406C8F33A40D-3f033c2b as the JMRI Node identity [main]
2017-05-12 18:36:36,839 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-6]
2017-05-12 18:36:36,946 configurexml.ConnectionConfigXml      INFO  - CAN Simulator Started [main]
2017-05-12 18:36:38,402 logix.WarrantPreferences              INFO  - Found Warrant preferences file: /Users/me/Library/Preferences/JMRI/My2/signal/WarrantPreferences.xml [main]
2017-05-12 18:36:38,422 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.util.ConcurrentModificationException
	at java.util.LinkedHashMap$LinkedHashIterator.nextNode(LinkedHashMap.java:711)
	at java.util.LinkedHashMap$LinkedEntryIterator.next(LinkedHashMap.java:744)
	at java.util.LinkedHashMap$LinkedEntryIterator.next(LinkedHashMap.java:742)
	at java.util.HashMap.putMapEntries(HashMap.java:511)
	at java.util.HashMap.<init>(HashMap.java:489)
	at jmri.implementation.SignalSpeedMap.setAppearances(SignalSpeedMap.java:296)
	at jmri.jmrit.logix.WarrantPreferences.loadSpeedMap(WarrantPreferences.java:303)
	at jmri.jmrit.logix.WarrantPreferences.openFile(WarrantPreferences.java:159)
	at jmri.jmrit.logix.WarrantPreferences.initialize(WarrantPreferences.java:655)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:258)
	at jmri.implementation.JmriConfigurationManager.lambda$load$473(JmriConfigurationManager.java:181)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:180)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:168)
	at apps.Apps.<init>(Apps.java:261)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)
	at apps.PanelPro.PanelPro.main(PanelPro.java:104)
```

A second try (after force quitting PP) succeeds.
Warrant preferences xml file attached:
[WarrantPreferences.xml.txt](https://github.com/JMRI/JMRI/files/997396/WarrantPreferences.xml.txt)

bobjacobsen, 2017-05-12:
I've not seen that, but I've seen "on wrong thread" warnings from the Warrant code. Are you also seeing those?

Are you using Warrants?  They seem to be doing a bunch of initialization even when not in use. We need to fix the threading problem(s), no question, but it would also be good to  have Warrants only initialize when in use.

silverailscolo, 2017-05-12:
I do not use Warrants, so they might just show default settings. Only experimented once or twice with the Dispatcher menu and the Transits table.

petecressman, 2017-05-24:
Please give an instance where Warrants initializes when not in use.

silverailscolo, 2017-05-26:
@petecressman Today again, when compiling based on current master and some, I see Warrant Preferences being loaded and cause a CME:
```
panelpro
runtime-library-selection
arch.lib.path /Users/egbertbroerse/Documents/Egbert/Computers/IntelliJ local/JMRI/lib/macosx/x86_64:/Users/egbertbroerse/Documents/Egbert/Computers/IntelliJ local/JMRI/lib/macosx
-normal-jmri-application
Launch normally (no debugger support)
2017-05-26 21:21:22,692 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-05-26 21:21:25,338 util.Log4JUtil                        INFO  - This log is appended to file: /Users/egbertbroerse/Library/Preferences/JMRI/log/messages.log [main]
2017-05-26 21:21:25,339 util.Log4JUtil                        INFO  - This log is stored in file: /Users/egbertbroerse/Library/Preferences/JMRI/log/session.log [main]
2017-05-26 21:21:25,432 apps.Apps                             INFO  - PanelPro version 4.7.4ish+egbertbroerse+20170526T1921Z+Rd3b52d328d starts under Java 1.8.0_66 on Mac OS X x86_64 v10.12.5 at Fri May 26 21:21:25 CEST 2017 [main]
2017-05-26 21:21:27,815 apps.Apps                             INFO  - Starting with profile My2.3f033c2b [main]
2017-05-26 21:21:28,303 node.NodeIdentity                     INFO  - Using jmri-406C8F33A40D-3f033c2b as the JMRI Node identity [main]
2017-05-26 21:21:29,942 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-6]
2017-05-26 21:21:30,013 configurexml.ConnectionConfigXml      INFO  - CAN Simulator Started [main]
2017-05-26 21:21:30,639 logix.WarrantPreferences              INFO  - Found Warrant preferences file: /Users/egbertbroerse/Library/Preferences/JMRI/My2/signal/WarrantPreferences.xml [main]
2017-05-26 21:21:30,664 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.util.ConcurrentModificationException
	at java.util.LinkedHashMap$LinkedHashIterator.nextNode(LinkedHashMap.java:711)
	at java.util.LinkedHashMap$LinkedEntryIterator.next(LinkedHashMap.java:744)
	at java.util.LinkedHashMap$LinkedEntryIterator.next(LinkedHashMap.java:742)
	at java.util.HashMap.putMapEntries(HashMap.java:511)
	at java.util.HashMap.<init>(HashMap.java:489)
	at jmri.implementation.SignalSpeedMap.setAppearances(SignalSpeedMap.java:296)
	at jmri.jmrit.logix.WarrantPreferences.loadSpeedMap(WarrantPreferences.java:303)
	at jmri.jmrit.logix.WarrantPreferences.openFile(WarrantPreferences.java:159)
	at jmri.jmrit.logix.WarrantPreferences.initialize(WarrantPreferences.java:655)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:258)
	at jmri.implementation.JmriConfigurationManager.lambda$load$473(JmriConfigurationManager.java:181)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:180)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:168)
	at apps.Apps.<init>(Apps.java:261)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)
	at apps.PanelPro.PanelPro.main(PanelPro.java:104)
```

petecressman, 2017-05-31:
My guess at what is happening is:
WarrantPreferences has the user's specifications for signal speed names and values.  WarrantPreferences  along with all  other preferences are loaded are loaded after the load of panel(s) has begun. 
BlockManager, DefaultSignalMastLogicManager and other signal system objects and open SignalSpeedMap to get speed information for panel objects. They load from a thread via ConfigXmlManager.loadOnSwingThread.

 SignalSpeedMap loads default signal speed data from the xml file, xml/signals/signalSpeeds.xml.  This data updated to the user's preference by WarrantPreferences  running on a main thread.  It seems to me that Preferences should be loaded before starting to load panels.  WarrantPreferences takes. as it's defaults xml/signals/signalSpeeds.xml

rhwood, 2017-05-31:
The log shows that WarrantPreferences is the very second preferences set loaded (the first being the Connections; every other set of preferences waits on those to be setup first), so no panels have been touched at this point.

petecressman, 2017-05-31:
My traces show that when a panel is loaded using the Start up Preference of "Open Panel file", the managers such as DefaultSignalMastLogicManager and BlockManager run on the swing thread before WarrantPreferences can begin.  If no panel is chosen, then as you say, none is loaded until a menu is picked.

closed 2017-06-06 by silverailscolo