JMRI/JMRI #3507: DefaultConditionalManager.getWhereUsedMap() does not work as documented

rhwood, 2017-05-16:
This method is documented as returning a copy of a HashMap, but returns the original HashMap itself. Which is correct, the method or the documentation?

    /**
     * Return a copy of the entire map.  Used by
     * {@link jmri.jmrit.beantable.LogixTableAction#buildWhereUsedListing}
     * @since 4.7.4
     */
    @Override
    public HashMap<String, ArrayList<String>> getWhereUsedMap() {
        return conditionalWhereUsed;
    }

Generally, unless you want the calling object to be able to directly manipulate the internals of the DefaultConditionalManager, it is preferable to pass a copy of an exposed collection instead of the collection itself.

KenC57, 2017-05-16:
I would think the comment was more of what was intended. At least if I write
a comment like that during the production, it was what I planned.

Looking at the code that calls that (only one place), it makes a copy of the
keySet() and works from that. So, as used today, it shouldn't matter. So I'd
make it match the comment.

-ken c

dsand47, 2017-05-16:
Yes, a **copy** was intended.  While the current use is read only, I don't want future uses having update access except using the supplied methods.  I am still learning Java and references.

The solution appears to be a new HashMap followed by putAll() and return the new map.

Dave Sand

rhwood, 2017-05-17:
> The solution appears to be a new HashMap followed by putAll() and return the new map.

Its actually simpler than that: ```return new HashMap<>(conditionalWhereUsed);```

Will be in #3509.

dsand47, 2017-05-17:
Yes, much simpler.  I took another look at the HashMap Javadoc and I now see the solution.  Thanks.

commit 8ef8ab0f (2017-05-17): Merge pull request #3509 from rhwood/javadoc-warnings

Minor Javadoc and other warnings cleanup
Closes #3507
closed 2017-05-17 by rhwood