JMRI/JMRI #3512: 4.7.4 Release Note - PR #3477

dsand47, 2017-05-17:
Based on my research of the side effects of PR #3477 on 4.6 using a 4.7.4 xml file, the release note text should be modified.

It currently states "_may not be readable by versions **before** JMRI 4.6_".  It should say before **_this_** version.

Known issues:
- Blocks are not created during loading.   If there are related layout blocks, those blocks will be created but  block specific data, such as occupancy sensors, is lost.
- Sensor groups cannot be added, changed or deleted.  A java exception occurs since the sensor group name does not exist.
- Logix and conditional user names are lost.
- Route user names are lost.

Note:   Routes, Logix and sensor groups still function since internally they use system names.

I should have included the release note concern in issue #3510.

Dave Sand

bobjacobsen, 2017-05-17:
You might have been looking at an older version of the release note.  Since [JMRI/website/eaaf64](https://github.com/JMRI/website/commit/eaaf64a8e94dbe40bf630fb2b35d60698a514993) it's been saying:

> This version of JMRI (and all later) writes a slightly streamlined version of panel file contents. Older files should still be readable by this version of JMRI, but files written by this version may not be readable by versions before JMRI 4.7.4 depending on their contents.

dsand47, 2017-05-17:
Dang browser cache.  Sorry.

closed 2017-05-18 by dsand47

bobjacobsen, 2017-06-04:
PR #3605 improves this.