JMRI/JMRI #3525: Travis CI times out on AboutDialogTest

rhwood, 2017-05-22:
The Travis CI tests for the AboutDialogTest are intermittently timing out.

bobjacobsen, 2017-05-22:
Just took a quick look at this. Two thoughts:

1) (Not a Jemmy expert, so this might not be a real issue) One test in the `AboutDialogTest` class includes:
```
        new Thread(() -> {
            // constructor for jdo will wait until the dialog is visible
            JDialogOperator jdo = new JDialogOperator(Bundle.getMessage("TitleAbout", jmri.Application.getApplicationName()));
            jdo.getTimeouts().print(System.err);
            jdo.close();
        }).start();
        dialog.setVisible(true);
        JUnitUtil.waitFor(() -> {
            return !dialog.isVisible();
        }, "About dialog did not close");
```
Is the threading on that right?  Looks like `close()` is being called on a non-Swing thread.  Or does Jemmy handle that automatically?

2) AboutDialog itself does some work while being set up. In particular,
```
        // add listener for Com port updates
        for (ConnectionConfig conn : InstanceManager.getDefault(ConnectionConfigManager.class)) {
            if (!conn.getDisabled()) {
                pane1.add(new ConnectionLabel(conn));
            }
        }
```
uses the `ConnectionLabel` that's being worked on in #3466. I don't see a possible source for a hang there, though.

pabender, 2017-05-22:
Sent from my iPhone

> On May 22, 2017, at 11:11 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> Just took a quick look at this. Two thoughts:
> 
> (Not a Jemmy expert, so this might not be a real issue) One test in the AboutDialogTest class includes:
>         new Thread(() -> {
>             // constructor for jdo will wait until the dialog is visible
>             JDialogOperator jdo = new JDialogOperator(Bundle.getMessage("TitleAbout", jmri.Application.getApplicationName()));
>             jdo.getTimeouts().print(System.err);
>             jdo.close();
>         }).start();
>         dialog.setVisible(true);
>         JUnitUtil.waitFor(() -> {
>             return !dialog.isVisible();
>         }, "About dialog did not close");
> Is the threading on that right? Looks like close() is being called on a non-Swing thread. Or does Jemmy handle that automatically?
> 
Jemmy handles that for us.

My understanding is that the JDialogOperator.close() method is effectively clicking the close button in the window decorations.

Paul

pabender, 2017-05-22:
There are two places ( in two different threads ) where the thread waits:

The constructor for just waits for a window with the correct name to be visible.  If that never happens, then this test will wait forever.

There is a JUnitUtil wait to see if the window is still visible.  This is checking the dialog object created by the test.

The main problem with Jemmy ( and JFCUnit ) is that because we are searching for the windows by title, it is actually possible to select any window on the screen with the right title, not just the window created in a specific test.  This means if there is already a window with the specified name visible, that window may be selected instead of the one we are looking for.

In this particular test class, the consequence is that if there is already a dialog with the right title visible, the constructor will return a JDialogOperator referencing a different window object than the one created by the class.p, so the second wait could never return, even if the first jdo.close operation happens.

Paul

bobjacobsen, 2017-05-22:
I don't understand the combination of these two lines:
> The constructor for just waits for a window with the correct name to be visible. If that never happens, then this test will wait forever.
> 
> There is a JUnitUtil wait to see if the window is still visible. This is checking the dialog object created by the test.

It's true that the spawned thread won't terminate, but that shouldn't keep the test from ending.  

The test for the window closing should time out quite rapidly. `JUnitUtil.waitFor(..)` should wait for at most `WAITFOR_MAX_DELAY = 15000` milliseconds.

pabender, 2017-05-22:
Sent from my iPhone

> On May 22, 2017, at 2:25 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> I don't understand the combination of these two lines:
> 
> The constructor for just waits for a window with the correct name to be visible. If that never happens, then this test will wait forever.
> 
> There is a JUnitUtil wait to see if the window is still visible. This is checking the dialog object created by the test.
> 
> It's true that the spawned thread won't terminate, but that shouldn't keep the test from ending.
> 
That should be true, but we are working with a modal dialog box ( see PR #2989, which added the testShowAndClose test ).
> The test for the window closing should time out quite rapidly. JUnitUtil.waitFor(..) should wait for at most WAITFOR_MAX_DELAY =
> 
> 15000 milliseconds.
> 
I think the problem may be we never get to this line.  

Think about it this way.  If the constructor for jdo finds an open window with the correct name, then jdo.close() happens.  If this happens, but the dialog is not the one we create for this test, the new dialog we create in this thread stays open, and the waitFor never actually executes.

So how do we fix that?

Here are some options:

1) make sure that there aren't any open windows with the same name as this one.  It is possible there is another test creating, and setting visible, a window with the same title.  I would suspect an action test.

2) make sure there are no open windows at all when we  we start the tests ( Jemmy doesn't care if the window is actually disposed, only that it is not visible.  I believe JFCUnit works in a similar manner.)

3) put a time limit on the test itself.  I am not sure this will be any more effective than the wait for however.

4) we could use a static reference to the dialog box for all tests in the class, setting it up in a BeforeClass method.  ( there are only two tests in the test class, a constructor test and a show/close test.  The constructor test never sets the frame visible )

I am inclined to try the 4th option first to see if that helps, but I really think there is another test class creating ( and displaying) a window with the same name.

Paul

bobjacobsen, 2017-05-22:
> If the constructor for jdo finds an open window with the correct name, then jdo.close() happens. If this happens, but the dialog is not the one we create for this test, the new dialog we create in this thread stays open, and the waitFor never actually executes.

I don't understand that, sorry. 

The thread is launched and does what it does. 

Meanwhile, the main test thread runs to the waitFor. If that's not satisfied, it should time out and report an error. 

How do we not get to the waitFor on the main test thread due to another window?

I think the problem is elsewhere, perhaps in the AboutDialog ctor. 

As a debug step, we could dump the list of Windows at the start of this test and see if there's one with a likely name. 

Bob

rhwood, 2017-05-22:
Note that until a Window is disposed, it will not be garbage collected, even if closed, unless dispose() is called on a window, it can still be used (java.awt.Window maintains a strong reference to every Window that is released only by dispose(), so even if the tests release all references, but dispose is not called, the Window remains available.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2017-05-22:
Sent from my iPad

> On May 22, 2017, at 3:53 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> 
> > If the constructor for jdo finds an open window with the correct name, then jdo.close() happens. If this happens, but the dialog is not the one we create for this test, the new dialog we create in this thread stays open, and the waitFor never actually executes.
> 
> I don't understand that, sorry. 
> 
> The thread is launched and does what it does. 
> 
> Meanwhile, the main test thread runs to the waitFor. If that's not satisfied, it should time out and report an error. 

This is where I think we have a problem.

The main thread creates the thread that creates jdo, and jdo does its thing.

The main thread then calls setVisible on dialog, because jdo can't do its thing until dialog is visible.

Since dialog is a modal dialog window, the main thread waits after the setVisible until the dialog is dismissed.

Then the waitfor is called and we move along.

> How do we not get to the waitFor on the main test thread due to another window?

Imagine what happens in the sequence above if there is already a visible window with the same name on the screen when jdo is created.  jdo will go ahead and do its thing, finding the open window, and closing it as required.  By the time we get to the setVisible on dialog, jdo is already finished with its work, so the modal dialog, which is now visible, stops further execution in the thread.

( incidentally, I am not speculating this can happen.  When I originally created this, we had a lot of frames named "Test Frame".  The test would just find one of those and close it. )

> I think the problem is elsewhere, perhaps in the AboutDialog ctor. 
> 
> As a debug step, we could dump the list of Windows at the start of this test and see if there's one with a likely name. 

We could.

I'll work on that tonight.

Paul

rhwood, 2017-05-22:
There is a method in JUnitUtil that closes all windows named resetWindows() that does that.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2017-05-22:
> Since dialog is a modal dialog window, the main thread waits after the setVisible until the dialog is dismissed.

Aha! This is probably the thing I missed. Thank you. 

I was thinking that the test thread was separate from the Swing event thread.  If it's actually running tests on the Swing thread, I can see how a modal dialog would mess stuff up. 

Thanks again.

pabender, 2017-05-24:
This should have been taken care of with #3531 and #3540

rhwood, 2017-06-06:
Have not seen any further instances of this.

closed 2017-06-06 by rhwood