JMRI/JMRI #3526: Frequent schema validation errors

rhwood, 2017-05-22:
All CI servers are repeatedly, but intermittently, failing with:
```
[ERROR] jmri.configurexml.LoadAndStoreTest.loadLoadStoreFileCheck[java/test/jmri/configurexml/load/BlockManagerXmlTest-4.7.4.xml (pass=true)](jmri.configurexml.LoadAndStoreTest)
[ERROR]   Run 1: LoadAndStoreTest>LoadAndStoreTestBase.loadLoadStoreFileCheck:173->LoadAndStoreTestBase.checkFile:134 expected:<    <memory value="[10:26 P]M">> but was:<    <memory value="[8:57 A]M">>
[ERROR]   Run 2: LoadAndStoreTest>LoadAndStoreTestBase.tearDown:190 Unexpected ERROR or FATAL messages emitted
```
or with
```
[junit] Testcase: loadLoadStoreFileCheck[java\test\jmri\configurexml\load\LoadFileTest277-4.7.4.xml (pass=true)](jmri.configurexml.LoadAndStoreTest):	Caused an ERROR
    [junit] expected:<    <memory value="1[2:01] PM">> but was:<    <memory value="1[1:33] PM">>
    [junit] org.junit.ComparisonFailure: expected:<    <memory value="1[2:01] PM">> but was:<    <memory value="1[1:33] PM">>
    [junit] 	at jmri.configurexml.LoadAndStoreTestBase.checkFile(LoadAndStoreTestBase.java:134)
    [junit] 	at jmri.configurexml.LoadAndStoreTestBase.loadLoadStoreFileCheck(LoadAndStoreTestBase.java:173)
    [junit] 
    [junit] 
    [junit] Testcase: loadLoadStoreFileCheck[java\test\jmri\configurexml\load\LoadFileTest277-4.7.4.xml (pass=true)](jmri.configurexml.LoadAndStoreTest):	Caused an ERROR
    [junit] Unexpected ERROR or FATAL messages emitted
    [junit] java.lang.AssertionError: Unexpected ERROR or FATAL messages emitted
    [junit] 	at apps.tests.Log4JFixture.tearDown(Log4JFixture.java:34)
    [junit] 	at jmri.configurexml.LoadAndStoreTestBase.tearDown(LoadAndStoreTestBase.java:190)
```

The specific test file and values vary, but its is always failing on a memory value with a time.

rhwood, 2017-05-22:
These failures appear to have been introduced in #3477

KenC57, 2017-05-22:
I see that both of those cases are the memory value IMCURRENTTIME. Any
chance the clock is running while this test is active?? If it ticked, that
would update the object and the time changes from what was in the file that
was loaded.

 

So timing could be much of the time when the test executes, the value is
loaded and tested before the minute changes. But if the test happens right
around the minute, the load would happen but then the minute ticks, updates
the value, and the test reads the updated value and fails.

-ken c

rhwood, 2017-05-22:
Prior to #3477, the test explicitly excluded the IMCURRENTTIME memory from tests, but #3477 moved the system name for the memory item to exclude to a different line than the changing value.

bobjacobsen, 2017-05-22:
#3529 should be a test and a fix.

closed 2017-05-22 by bobjacobsen