JMRI/JMRI #3536: NxWarrant fails occasionally in Travis CI

bobjacobsen, 2017-05-23:
Typical traceback:

```
[ERROR] Tests run: 3, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 4.482 s <<< FAILURE! - in jmri.jmrit.logix.NXFrameTest
[ERROR] testNXWarrant(jmri.jmrit.logix.NXFrameTest)  Time elapsed: 4.093 s  <<< FAILURE!
java.lang.AssertionError: Num Blocks in Route expected:<7> but was:<5>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:645)
	at jmri.jmrit.logix.NXFrameTest.testNXWarrant(NXFrameTest.java:124)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at junit.extensions.jfcunit.JFCTestCase$5.run(JFCTestCase.java:580)
	at junit.extensions.jfcunit.JFCTestCase$3.run(JFCTestCase.java:478)
	at java.lang.Thread.run(Thread.java:745)
[INFO] Running jmri.jmrix.sprog.update.SprogHexFileTest
```

rhwood, 2017-05-24:
I think the fix for this has been merged into the master branch; will need to monitor to verify.

petecressman, 2017-05-24:
I ran the logix package over 375 times (i.e. more than 20,000 individual tests) and did not get this error.  However, i did notice the following error when the package was run in batches of 75 or more. (batches of 50 or less did not have the error)

        [java] ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main] jmri.util.exceptionhandler.UncaughtExceptionHandler.?()
        [java] java.lang.NullPointerException: Required nonnull default for jmri.jmrit.logix.WarrantPreferences does not exist.
        [java] 	at java.util.Objects.requireNonNull(Unknown Source)
        [java] 	at jmri.InstanceManager.getDefault(InstanceManager.java:182)
        [java] 	at jmri.implementation.SignalSpeedMap.lambda$2(SignalSpeedMap.java:84)
        [java] 	at jmri.InstanceManager.lambda$0(InstanceManager.java:352)
        [java] 	at java.util.HashMap$KeySpliterator.forEachRemaining(Unknown Source)
        [java] 	at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
        [java] 	at jmri.InstanceManager.notifyPropertyChangeListener(InstanceManager.java:351)
        [java] 	at jmri.InstanceManager.setDefault(InstanceManager.java:292)
        [java] 	at jmri.jmrit.logix.WarrantPreferences.lambda$0(WarrantPreferences.java:132)
        [java] 	at java.util.Optional.orElseGet(Unknown Source)
        [java] 	at jmri.jmrit.logix.WarrantPreferences.getDefault(WarrantPreferences.java:131)
        [java] 	at jmri.implementation.SignalSpeedMap.lambda$0(SignalSpeedMap.java:55)
        [java] 	at java.beans.PropertyChangeSupport.fire(Unknown Source)
        [java] 	at java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)
        [java] 	at java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)
        [java] 	at jmri.beans.Bean.lambda$6(Bean.java:154)
        [java] 	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        [java] 	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        [java] 	at java.awt.EventQueue.access$500(Unknown Source)
        [java] 	at java.awt.EventQueue$3.run(Unknown Source)
        [java] 	at java.awt.EventQueue$3.run(Unknown Source)
        [java] 	at java.security.AccessController.doPrivileged(Native Method)
        [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        [java] 	at java.awt.EventQueue.dispatchEvent(Unknown Source)
        [java] 	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        [java] 	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        [java] 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        [java] 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        [java] 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        [java] 	at java.awt.EventDispatchThread.run(Unknown Source)

rhwood, 2017-06-06:
Have not observed this recently.

closed 2017-06-06 by rhwood