JMRI/JMRI #3537: Trailing blanks in Sensor Username problem

devel-bobm, 2017-05-23:
FYI - with 4.7.4ish code, it is possible to use the "Add" feature of the "Sensor Table" to add an Internal Sensor which has a trailing blank in the "User Name".  It is accepted and is kept as part of the user name.

But it is currently impossible to use that "User Name" with a trailing blank in the "System/User Name" of an Antecedent Variable of a Logix conditional or in the "System/User Name" field of a consequent Action.  When such is used, with the trailing space, the Logix Conditional reports that the name, _without_ the trailing space, does not match an existing sensor.  The logix conditional is not updated.

If the Internal Sensor is "Rename"d so the User Name no longer has the trailing blank, the problem disappears.

This problem presumably appears for "actual" Sensors, too.

If using the "Pick List", it may be necessary to exit the "Pick List" and exit the "Logix Conditional" and then re-edit the "Logix Conditional" to get a Pick List which has the updated "User Name" for the Sensor. 

Ref: #2535

geowar1, 2017-05-23:
MOOF! I'll take a look… (someone with privileges please assign?)

geowar1, 2017-05-23:
See Pull Request #3538

bobjacobsen, 2017-05-23:
We've been over this. A large amount of work went into infrastructure to fix this _once_, instead of scattered across all the various input and output places.  We should actually _use_ that infrastructure.

See the [documentation on this](http://jmri.org/help/en/html/doc/Technical/Names.shtml#programmers)

The right solution is to turn that on, which is a small change in [NamedBean.html#normalizeUserName](http://jmri.org/JavaDoc/doc/jmri/NamedBean.html#normalizeUserName-java.lang.String-) and NamedBeanTest.

This would remove trailing spaces silently throughout.

But the consensus (which I didn't favor) was _not_ to turn that on until 
1. we have more the places where people can do input covered so that the non-fixable errors (which throw `NamedBean.BadUserNameException`) would do something useful in the user interface instead of just logging the `BadUserNameException` (or maybe really messing up, depending on how the `BadUserNameException` was caught)
2. Something Was Done to handle cases where trailing spaces were actually meaningful when reading existing files.  (This is the part I disagree with; I don't think we can do much other than logging those and fixing them manually, but others wanted to work on this)

I think the right thing to do for JMRI 4.7.4 is to turn this feature on, which will get it across the various NamedBean managers.

geowar1, 2017-05-24:
Looking thru just the "java/src/jmri/jmrit/display/layoutEditor/…" source I'm finding a lot of `.trim()`'s… In some places it's just cleaning up numeric entry (like for angles, rotations, etc.); no harm- no foul. But then there's all the signal/sensor code… and I don't know if those names are user names or system names… so I'm not going to touch them for now… but to fix _THIS_ issue I've "done the right thing" (™bobjacobsen) and fixed the appropriate code to use the `normalizeUserName` method.

devel-bobm, 2017-05-24:
Jenkins build number 1555 includes these fixes and does not show the problem I reported.

closed 2017-05-24 by devel-bobm