JMRI/JMRI #3548: AbstractMRTrafficController and subclasses do not terminate threads

rhwood, 2017-05-25:
The ```terminate()``` method of AbstractMRTrafficController is not always called by subclasses, preventing tests from closing any threads that were opened by the AbstractMRTrafficController (specifically the Z21XNetPacketizer overrides terminate() to do nothing).

pabender, 2017-05-25:
The terminate() method in AbstractMRTrafficController is not about cleaning up threads.

The terminate method ( which is called by the shutdown hook ) is about making sure the connected system ends up in a consistent state. When terminated.  In particular, it makes sure the system is no longer in service mode.

In the specific case of the Z21XNetPacketizer, the method does nothing because it is a child of another connection.  The Z21TrafficController DOES allow the standard terminate routine to run.  Running that in both places is  it required.

The terminate method NEVER has performed any action to shit down threads.  This is not a bug.

rhwood, 2017-05-25:
The fact that there is no mechanism to terminate threads spawned by AbstractMRTrafficController or its subclasses other than killing the JVM remains a bug, even if that needs to be done outside the terminate() method. It definitely needs to be in the finalizer and ideally in an exposed mechanism so that, for example, the persistent failures in tests because these threads are not being killed can be prevented.

pabender, 2017-05-25:
@rhwood,

I am having trouble conceptualizing how this can be an issue.

We typically have a process by which we reset the instance manager, then run our tests, then reset the instance manager again before moving on.

Once we reset the instance manager and the test class exits, we loose all reference to any objects created by that test class.  We have no way to send output to or receive input from the traffic controller, so even if the threads still exist, they will perform not work.

bobjacobsen, 2017-05-25:
I think there probably should be a mechanism for shutting down the `AbstractTrafficController` subclass instances that includes ending (or at least attempting to end) their threads. (Not every system uses a subclass of that, but enough do that it would be progress for the tests)

Some design thoughts:

- Should it be built into the `AbstractTrafficController`, which is in the main code, or into the various `*Scaffold` test classes, e.g. `java/test/jmri/jmrix//cmri/serial/SerialTrafficControlScaffold.java`?  `Thread.stop()` is [deprecated](https://docs.oracle.com/javase/8/docs/api/java/lang/Thread.html#stop--) and I'm not sure we should be adding it to the main code. I care less about the test code for that.  Ensuring that we have a more robust and modern structure for stopping these threads doesn't seem like a priority for me.  But of course, others might want to do that. And doing it in the test classes (1) is more work, as there are at least 16 of them and (2) might still not get full coverage.

- In any case, I'm _not_ fond of adding internal clean up (like forcing a GC or stopping threads) when the program is terminating. Saving state, making sure external hardware is clean; those are fine.  But adding work to the JMRI shutdown that will be automatically done (if coarsely) by the JVM just adds delay.  So regardless of where it lies, I'd like "thread shutdown" to be separate from the existing `terminate()` call.  To put it another way:
   - `terminate()` is for essential operations at the end of the program
   - `cleanUpForRunningEGForTests()` is a separate call

I'd be happy to work on adding this to the `*TrafficControlScaffold*` test-support classes this weekend unless somebody wants to do another approach.

bobjacobsen, 2017-05-25:
> We typically have a process by which we reset the instance manager, then run our tests, then reset the instance manager again before moving on.
>
> Once we reset the instance manager and the test class [...]

We (in the sense of the main line of test code) lose access, but any other threads that still exist don’t necessarily lose their access.  And any objects those (still live) threads have references to are kept around, not GC’d, and can be invoked if/when something happens on the thread.

In e.g. NCE, that chain of references can go all the way back to the NceSensorManager and NceTurnoutManager instances, perhaps further.

That said, I don’t know why those threads might be doing something that causes test hangs.  They do have internal timeouts, but I don’t see those doing anything other than issuing some annoying Warn messages.

rhwood, 2017-05-25:
Whatever process that spins out that Z21whatever thread that then logs missing manager errors all over subsequent tests after the InstanceManager is reset and is no longer holding the defaults the Z21whatever expects has been causing false errors in tests for a long time. It's clearly not terminating threads it creates. That's the immediate issue.

An enhancement that could come from this is that disabling (not removing) a connection could be immediate instead of only on restart.

Randall Wood
On my phone

pabender, 2017-05-25:
> On May 25, 2017, at 10:14 AM, Randall Wood <notifications@github.com> wrote:
> 
> Whatever process that spins out that Z21whatever thread that then logs missing manager errors all over subsequent tests after the InstanceManager is reset and is no longer holding the defaults the Z21whatever expects has been causing false errors in tests for a long time. It's clearly not terminating threads it creates. That's the immediate issue.

I have no clue what you are talking about.  I haven't seen any Z21 errors on the CI servers for a long time.

Paul

rhwood, 2017-05-25:
https://github-production-repository-file-5c1aeb.s3.amazonaws.com/42789514/1028737?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20170525%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170525T144205Z&X-Amz-Expires=300&X-Amz-Signature=20aa8a204cbfe68cf184539d020b4cb8437e75b12400d96e19e45d8c70562633&X-Amz-SignedHeaders=host&actor_id=297232&response-content-disposition=attachment%3Bfilename%3D7624-2.txt&response-content-type=text%2Fplain shows a typical Z21 failure appearing as a failure in another test.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2017-05-25:
@rhwood that link doesn't work

bobjacobsen, 2017-05-25:
I've attached the file that was linked.
[7624-2.txt](https://github.com/JMRI/JMRI/files/1029328/7624-2.txt)

pabender, 2017-05-25:
Ok, so the problem is actually that we reset the ConnectionConfig manager to null, and it's supposed to be a non-null object, so we don't check anywhere to see that it is null.

Yes, there may be a thread hanging around, but the but here is that the reset in the tests violates the intention of the NonNull default.

rhwood, 2017-05-25:
We don't set the ConnectionConfigManager to null anywhere, we do reset the InstanceManager, but since the overwhelming majority of unit tests do not need a ConnectionConfigManager to function correctly, we don't need to retain it, except for tests that spawn threads and don't close them when tearing down. I'd rather the test exit cleanly than have other tests retain or create unneeded objects for those tests.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2017-05-25:
We recommend that code _not_ cache the `InstanceManager` instance, but rather pick up a new one from the static variable each time.  That means a long-running test, e.g. remanent thread, isn't going to be getting what it's thinking it's getting once the test is over and another has started.  Better to have a clean end to each test by forcing an end to threads (at least as much as possible).

pabender, 2017-05-25:
Ok. See #3550

This follows the recommendation from oracle on terminating threads (since Thread.stop() is deprecated).

bobjacobsen, 2017-08-17:
All of the threads launched by AbstractMRTrafficController itself are now (AFAIK) closing.  There may be (probably are) further threads launched downstream that aren't, but I think that's a separate issue.  So I'm closing this.

closed 2017-08-17 by bobjacobsen