JMRI/JMRI #3565: Operation standalone test fails, dependency on WebServerPreferences

DanielBoudreau, 2017-05-28:
I'm now seeing an exception when running just the operation tests.

There was 1 error:
1) testXMLFileCreate(jmri.jmrit.operations.setup.OperationsSetupTest)java.lang.NullPointerException: Profile must be specified.
	at java.util.Objects.requireNonNull(Objects.java:228)
	at jmri.beans.PreferencesBean.<init>(PreferencesBean.java:36)
	at jmri.web.server.WebServerPreferences.<init>(WebServerPreferences.java:148)
	at jmri.web.server.WebServerPreferences.lambda$0(WebServerPreferences.java:155)
	at java.util.Optional.orElseGet(Optional.java:267)
	at jmri.web.server.WebServerPreferences.getDefault(WebServerPreferences.java:154)
	at jmri.jmrit.operations.setup.Setup.store(Setup.java:1789)
	at jmri.jmrit.operations.setup.OperationsSetupXml.writeFile(OperationsSetupXml.java:63)
	at jmri.jmrit.operations.setup.OperationsSetupTest.createTestSetup(OperationsSetupTest.java:451)
	at jmri.jmrit.operations.setup.OperationsSetupTest.testXMLFileCreate(OperationsSetupTest.java:308)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at jmri.jmrit.operations.PackageTest.main(PackageTest.java:22)

Here's the code:

   public static Element store() {
        Element values;
        Element e = new Element(Xml.OPERATIONS);
        e.addContent(values = new Element(Xml.RAIL_ROAD));
        if (Setup.getRailroadName().equals(WebServerPreferences.getDefault().getRailRoadName())) {
            values.setAttribute(Xml.NAME, Xml.USE_JMRI_RAILROAD_NAME);
        } else {
            values.setAttribute(Xml.NAME, getRailroadName());
        }

The line:

 if (Setup.getRailroadName().equals(WebServerPreferences.getDefault().getRailRoadName())) {

is the problem.  It looks like I need to initialize a profile before running the test.  What is the correct procedure?

Dan

rhwood, 2017-05-28:
Add ```JUnitUtil.resetProfileManager();``` to the test's ```@Before``` annotated method (or to the ```setUp()``` if using JUnit3 tests).

DanielBoudreau, 2017-05-28:
does the trick.  thanks,

closed 2017-05-28 by DanielBoudreau