JMRI/JMRI #3574: Throttle button issue noticed in JMRI 4.7.4

bobjacobsen, 2017-05-29:
From a [jmriusers post](https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/138766):

> Update from 4.7.3 to this version. ... One thing I have noticed is that, should one open a throttle in JMRI and subsequently use the arrow keys to adjust speed, the selected radio button for forward or reverse changes as welll. Doesn't affect locomotive running direction, just means that the radio button that is selected may not necessarily correspond to locomotive direction.

Pugwash1, 2017-06-04:
I've just noticed it only happens after loading a Panel. Prior to that it works as expected.

Pugwash1, 2017-06-04:
No, thats not quite it, it starts after you change direction. It stops after you Idle or EStop.

Pugwash1, 2017-06-05:
using setFocusTraversalKeys to reset the forward and reverse keys did not work, the ButtonGroup does not support the method and placing it in the container panel did not work.

This works.        
        forwardButton.setFocusable(false);
        reverseButton.setFocusable(false);

bobjacobsen, 2020-01-08:
Fixed some time before 4.19.1

closed 2020-01-08 by bobjacobsen