JMRI/JMRI #3575: Layout Editor line width issue noticed in JMRI 4.7.4

bobjacobsen, 2017-05-29:
From a [jmriusers thread](https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/138760)
> Just installed the new 4.7.4 release. It seems to works. However it looks like something changed in my panel: When in edit mode the line width appears correct (6 main, 4 siding), but when i go to operation mode the sidings track width is very thin, like 1 pixel in width. It seems just a cosmetic issue.

A similar, perhaps related comment & update by Peter Ulvestad on [jmri-developers](https://sourceforge.net/p/jmri/mailman/message/35863982/):
> Using antiX Linux, everything appears to start okay in LocoNet simulator but there is an issue with panel.
>
>All the lines appear too big when panel starts but go to normal right away. There is also a crossover that when thrown, all the layout lines double in size. It only seems to happen with this one crossover. This also happened when I tried in Windows 10. 
> Panel attached, problem turnouts in top left(Lumby).

>Correction, it is only the Turnout circles that increase in size (must be old eyes).

Followed by Steve G's comment:

> Yes, the turnout circles are now drawn thicker for some reason even when no line going to it is a main.

geowar1, 2017-05-29:
I assume you're talking about Layout not Panel? If so this is probably me… (assign please?)

bobjacobsen, 2017-05-29:
It's probably Layout Editor, as Panel Editor and Control Panel Editor don't really have line widths.  But all I know is in those list postings.

bobjacobsen, 2017-05-29:
A bit more from jmriusers:
> More siding width trouble, plus turnout circles...
> On one of my Layout Panels, probably the first one I created, the Siding Lines are displaying 1 pixel wide at load but are defined as 2, and stay 1 pixel wide regardless of what I set them to. On my other layout panel (my staging tracks) they show up 2 pixels wide at load and change as I set them, but then change back to 1 pixel on their own after a minute or so and stay that way.
> The turnout circles are much thicker (not thinner) than they should be as well, on both.
> The lines that are drawn for the turnout legs are displaying and responding to siding width properly.
> My original layout panel just drew the sidings back to 2 pixels on its own. And now the 2nd panel has gone back to 1 pixel wide for siding tracks.
> Mainline track widths seem unaffected.

Pugwash1, 2017-05-30:
I noticed it on layout panel.
Steve G.

KenC57, 2017-05-31:
I just updated a CMRI/LocoNet layout to 4.7.4 and it works fine. But I too see that LE track that isn't 'mainline' or dotted look thinner than normal. The lines for turnouts and crossovers look thinner than I thought last version, but thicker than the normal track. This is a Win Vista system. I did a complete uninstall before doing the install.
-ken c

geowar1, 2017-05-31:
I did find and fix one bug with track segments widths… should be our next engineering build.

Play with setting the track widths to weird values (like 7 & 11 and vice-versa)… and see if any lines aren't being drawn at those widths. Grab screen shots if you can.

If you can mail those (annotated if possible) and your saved layout file to me (<geowar1@mac.com>) and I'll take a another look.

bobjacobsen, 2017-06-03:
See PR #3583

geowar1, 2017-08-11:
Can we confirm this as fixed and close?

closed 2017-08-11 by bobjacobsen