JMRI/JMRI #3578: "MS100" connections shown when not available

devel-bobm, 2017-05-30:
JMRI can only configure "MS100" interface hardware when using some more-modern O/Ses, but the option does show up - with a name `(LocoNet MS100 not available)`.  But the user can select this "non-option", and, when JMRI restarts, JMRI hangs without presenting a GUI.

Rather than presenting a useless option to the user, I think it makes sense to remove the option when JMRI identifies that the O/S doesn't support the MS100 hardware.

Thoughts?

bobjacobsen, 2017-05-30:
Could you try this with JMRI 4.7.4 (or the "release-4.7.4" branch)?

There was a _lot_ of special purpose MS100 code that's now been removed with the move from RXTX to PureJavaComm.  There might still be broken things, but that's the code base to look at now.

devel-bobm, 2017-05-30:
It appears (if I've gitted corectly) that the release-4.7.4 codebase presents the `LocoNet MS100` as a valid connection interface option on Win7-64.

Does this mean that PJC is able to generate the non-standard baud rate required by the MS100?  Can it do so across serial interfaces, including various USB-to-RS232 devices?

KenC57, 2017-05-30:
Bob M,

 

I think that the MS100 would be just one more device needing the tests.
Maybe it does support it, a test with real hardware is the only thing I'd
consider meaningful.

-ken c

bobjacobsen, 2017-05-30:
It's not at all clear whether the preferred 16600bps or fall-back 16457bps rates will be available on all hardware.  It works with a Keyspan HS USB-Serial adapter on my Mac with macOS 10.12.5 (and RXTX was never able to do that), but that's just one data point.  Real tests are going to be important.

But in any case, I don't think there are all that many _new_ users of the MS100 these days, so it's mostly a question for people with existing setups.

Our [MS100 web page](http://jmri.org/help/en/html/hardware/loconet/MS100.shtml) has lots of warnings, including

> We strongly recommend that new setups not use the MS100, and existing users plan on converting to a better solution before it becomes necessary.

That seems to still be true.  I'm not sure that this is still true:

> Because neither of these rates are available, the MS100 can't be used with Mac OS X, Windows 7 or most Windows Vista machines.

But it might make sense to leave that there, just in case, until we hear otherwise.

devel-bobm, 2017-06-02:
Another data point: MS100 with test release 4.7.4 with PJC (PureJavaComm) works on Win7-64 with an unknown USB-to-serial adapter which uses the same chipset as the LocoBuffer-II.  It was useless with previous non-PJC JMRI releases.

Assuming the move to PJC is permanent, this item can be closed.

closed 2017-06-13 by bobjacobsen