JMRI/JMRI #3588: SourceForge is killing our Issue Reporting tool

rhwood, 2017-06-01:
From an email:

> Dear SourceForge project owner,
>
> As a result of changing electronic messaging and privacy requirements
around the world, SourceForge will implement a number of changes that will
affect your project. Please review the following carefully and make
necessary changes before June 26, 2017.
>
> SourceForge will stop support for email sending from your Project Web space.
>
> That includes all transactional email from systems such a Wordpress,
Drupal, and similar. Since we also block outbound connections due to past
abuse (see
https://sourceforge.net/p/forge/documentation/Project%20Web%20and%20Developer%20Web/#outbound-connectivity),
it will not be possible to use email services such as MailChimp, and thus
no way to send email from your project web site.
>
> We deeply regret having to bring this functionality to end of life, but
since we cannot verify or control this type of email traffic going out from
the sourceforge.net domain, compliance with new electronic messaging and
privacy laws makes it impossible to continue the service.
>
> Please also refer to our full blog post at
https://sourceforge.net/blog/sourceforge-project-e-mail-policy-update/
If you have any questions and/or need further help making these adjustments
please email sfnet_ops@slashdotmedia.com.
>
> Thank you for your understanding and continued support,
>
> The SourceForge team

bobjacobsen, 2017-06-01:
I like the way that tool provides quick access to a comprehensive set of logs, context, profile, etc.  So I think it's worth keeping around.

Currently, the tool works with a PHP script at `http://jmri.org/problem-report.php`.

I'm happy to change the JMRI code to route to a different DNS name that points to a different machine.  Having it go through a name other than "jmri.org" gives us flexibility against later changes. But after the SF.net support goes away that'll only work for future JMRI versions (4.7.5 and later); older JMRI versions will break without a useful error message.

I think that we _might_ be able to use some form of redirection on the `jmri.org/problem-report.php` URL to get older versions to work, but that's not something I have a lot of expertise with.  Can `.htaccess` do this?  Can a PHP script emit a cross-domain redirect?

bobjacobsen, 2017-06-01:
Working on a differentt approach:  Have problem-report.php on jmri.org do a redirect (before sending mail) to another machine that can send mail.  This will continue to work with existing JMRI versions.

bobjacobsen, 2017-06-01:
I think this is now working with no changes to the distributed JMRI code, just to the infrastructure.  

The JMRI code is still invoking PHP at jmri.org, which redirects the POST request to another machine to do the actual mailing.  Our usual Jenkins-based server uploads will handle any changes needed from now on.

closed 2017-06-01 by bobjacobsen