JMRI/JMRI #3589: Signal Head NMRA Dcc Signal Decoder

Pugwash1, 2017-06-01:
The first 2 bytes are in a 10AAAAAA 0AAA0AA1 format, the 3rd byte is the signal colour. Notice the 2nd group of bits are 111 for address 1 thru 129 (rebased to 0 before sending) and then just silly. The comments state that we use the "alt" interpretation.
Address Data Sent
1       10000000 01110001 0 
2       10000000 01110010 0
3       10000000 01110101 0
5       10000001 01110001 0 
9       10000010 01110001 0 
17      10000100 01110001 0 
33      10001000 01110001 0 
65      10010000 01110001 0 
129     10100000 01110001 0 
257     10000000 01100001 0 
513     10000000 01010001 0 
1025    10000000 00110001 0
2048    10111111 00000111 0 

What I think it should be is
1       10000000 00000001 1 
3       10000000 00000101 1 
9       10000010 00000001 1 
17      10000100 00000001 1 
33      10001000 00000001 1 
65      10010000 00000001 1 
129     10100000 00000001 1 
257     10000000 00010001 1 
513     10000000 00100001 1 
1025    10000000 01000001 1 
2048    10111111 01110111 1

In NmraPacket.java (line ~601) we do "int highAddr = ((~boardAddr) >> 6) & 0x07;"
I believe it should be "int highAddr = (boardAddr >> 6) & 0x07;" 

I have not submitted a pull req. as I'm not confident I've got it right. If it is wrong someone should have noticed by now.

bobjacobsen, 2017-06-02:
> I believe that this is consistent with the code in accSignalDecoderPktCommon() and the QuadLN_S signaling has always worked fine with JMRI.

There are a bunch of test cases in the java/test/jmri/NmraPacketTest.java file that came from physical hardware.  I think a couple different types were used, so they should be pretty reliable examples.

> Perhaps Bob J will fill in the history of the address bit encoding (if it isn’t too painful!)

Long story, lots of details that maybe aren’t so interesting after all this time, but the bottom line is that nobody should assume that the NMRA packet formats make any kind of technical sense.  They weren’t defined that way.  So, even though it’s not a good thing, people should just get to used to things like arbitrarily having to add or subtract one, some fields inverted and others not, etc.

Pugwash1, 2017-06-03:
OK. Closing this.

closed 2017-06-03 by Pugwash1