JMRI/JMRI #3597: LocoNet Server is JMRI Version Specific

rhwood, 2017-06-02:
The LocoNet Server from version 4.6 back appears to work across multiple JMRI versions, but not with JMRI version 4.7.1 or newer. This is the error that was reported:
```
WARN  - sendLocoNetMessage: Exception: java.rmi.ServerException: RemoteException occurred in server thread; nested exception is: 
	java.rmi.UnmarshalException: error unmarshalling arguments; nested exception is: 
	java.io.InvalidClassException: jmri.jmrix.loconet.LocoNetMessage; local class incompatible
```

pabender, 2017-06-02:
@rhwood,

That is not a bug.  That is an indication that the parameters of the invoked method changed OR that one of the objects used as a parameter changed.  

In this particular case, something changed in the loconet message class.

In other words, the protocol changed, and you are trying to use two different versions of the protocol.

This isn't a bug, but there should be a note that recommends using the same JMRI version on both the client and server.

bobjacobsen, 2017-06-03:
This might be an inadvertent change. Checking.

pabender, 2017-06-03:
@bobjacobsen 

It might be due to removing the serialization, but if a field was changed, that could also cause the problem.

bobjacobsen, 2017-06-03:
#3600 is meant to fix this.  My tests with JMRI 4.6 and master-post-PR are OK; master-post-PR doesn't work with JMRI 4.7.1 through 4.7.4; will update release notes to that effect.

closed 2017-06-05 by bobjacobsen