JMRI/JMRI #3601: Zooming chops text strings in layout panel

bobjacobsen, 2017-06-04:
From email from a user:

> Situation, if you open the panel and set the zoom to No Zoom, you will notice the location  labels and turnout numbers are complete (Creamery, Bear Run Jct, etc). However if you take the zoom to 1.5, 2,0 or larger, part of the right end on some of the turnout numbers and part of the location labels begin to get trimmed.
>
> Ignore the number offset locations as I am working on XtrkcadReader to translate them to the proper positions but have not quite solved that problem.
>
> No sure  how this can be happening. I am including a single label xml file which I created in the Layout Editor and it does not trim the labels when zoomed.

with followup:

> Oh sorry, forgot to mention I am running JMRI 4.7.3 on a Windows 10 gaming maching with 24GB of ram and the latest JAVA JDK.

[JGS_Layout_v45-gaw.xml.txt](https://github.com/JMRI/JMRI/files/1050042/JGS_Layout_v45-gaw.xml.txt)
[oneLabel111.xml.txt](https://github.com/JMRI/JMRI/files/1050043/oneLabel111.xml.txt)

geowar1, 2017-06-04:
Assign me please? TIA.

geowar1, 2017-06-04:
I'm unable to reproduce on a PanelPro version 4.7.5ish+geowar+20170604T1851Z+R85798db640 build… Tried all different zoom levels and am not seeing any label clipping… in ether edit nor non-edit modes. Ask them to reproduce on a 4.7.5 build.
Can you ask them to send me (private email ok) screen shot?
Also, let them know that Track Segments in 4.7.5 support flex track (Bezier curves) with up to 8 control points (not counting the end points).

wolfsonga, 2017-06-05:
Okay, the file I sent you is named JGS_Layout_v45-gaw.xxx. Here is what happens when I rename a copy from ending in .xxx to ending in .xml and doing the following:
1. Open file in PanelPro from Panels -> Open Panels... 
2. Once up, turn off Edit Mode under Options
3. Set Zoom -> No Zoom
4. You will notice that labels such as Creamery, Bear Run Jct and all numbers are shown completely
5. Set Zoom -> x 6.0
6. You should no see that the end of the labels and some of the numbers are chopped off.

I am running under PanelPro 4.7.3+R202c9ee and when I go to a command prompt java -version reports it is 1.8.0_131 which I installed as part of installing the latest Oracle JDK.

If the attached image is not showing up at x 6.0, then it might be that the labels are being put on top and are not being covered by something adjacent that is invisible or something got  fixed between the version I am running and the one you are  using.
=gerry=
[image]

geowar1, 2017-06-05:
Could be… Or maybe there's just something different on macOS… Here's my screen shot at x6.0:
[image]

wolfsonga, 2017-06-06:
Hmm, I cleaned up some number and position things so here is a .xml file of
my latest translated version. Send me what you see on the diagonal text and
numbers when you zoom out to x 6.0. This is beginning to sound like a
library translation problem and may be outside of JMRI but I will pass some
of this to Randall and see if it triggers anything with him. Thanks,
=gerry=

geowar1, 2017-06-06:
>>here is a .xml file
No file attached… try again?

wolfsonga, 2017-06-06:
Oops, sorry,
=gerry=
PS I may have narrowed it down to being a font library sizing issue in
java. Let me know what results you get with the other xml file included.
(other files are for better understand and visualization of whats happening)

wolfsonga, 2017-06-06:
Hey George, just went back and looked at last sent mail and the one before
and both had attachments. Do I need a private address to get these
attachments to you? If so, what is it?
=gerry=
PS what time zone are you. I am east coast usa.

geowar1, 2017-06-06:
On Jun 6, 2017, at 9:27 AM, wolfsonga <notifications@github.com> wrote:
> Hey George, just went back and looked at last sent mail and the one before
> and both had attachments. Do I need a private address to get these
> attachments to you? If so, what is it?

I got the attachment to the 1st email but not the 2nd…
You may email me directly at my address below.

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

geowar1, 2017-06-06:
On Jun 6, 2017, at 9:24 AM, wolfsonga <notifications@github.com> wrote:
> PS I may have narrowed it down to being a font library sizing issue in
> java. Let me know what results you get with the other xml file included.
> (other files are for better understand and visualization of whats happening)

That’s very possible… I’m sure that there are differences between macOS & Window’s fonts.

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

wolfsonga, 2017-06-06:
Aah, I was able to get rid of the trim at the end by appending three blank
spaces and this gave enough space to prevent the clipping. However, there
is one thing in JMRI label strings that is really bad. If you rotate the
strings, they become fuzzy and pixelated but if you use orientation and us
vertical up or vertical down they are nice and clean. Is there any way to
add horizontal flip to this orientation set? The reason this is need is
that when making panels to be viewed from different aisle, the orientation
has to be changed on the tablet to either 90, 180 or 270. The two verticals
take care of reorienting the text for 90 and 270 but the "flip" (180) is
missing for standing on the opposite side of a layout peninsula and when
you  try and do the equivalent with rotate, the letters get all fuzzy and
pixelated (yuck).

Could you take a look at orientation and see if the 180 option can be added
easily.
Thanks,
=gerry=

wolfsonga, 2017-06-06:
PS Did you get all the files I sent?

geowar1, 2017-06-07:
On Jun 6, 2017, at 1:56 PM, wolfsonga <notifications@github.com> wrote:
> Aah, I was able to get rid of the trim at the end by appending three blank
> spaces and this gave enough space to prevent the clipping. However, there
> is one thing in JMRI label strings that is really [...]

I’ll have to check the sources…

> The reason this is need is that when making panels to be viewed from different
> aisle, the orientation has to be changed on the tablet to either 90, 180 or 270.
> The two verticals take care of [...]

I’m /afk until next week… I’ll have a look see then.

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

geowar1, 2017-06-30:
Hmmmmm… the "Orientation" for JLabel's is inherited from Java swing code… which doesn't recognize an "upside down" or "180°" option (these are "language reading orientations" (directions)). There may be a way to do this via transforms… I'll have to research further…
Note: I turned on anti-aliasing for macOS and Windows… this may help with the 180° rotation being "fuzzy and pixelated (yuck)". ;-)

wolfsonga, 2017-06-30:
If the anti-aliasing fixes the rotation fuzzy issue then finding a 180
upside down switch like the vertical up and vertical down ones become a
minor issue. It is only needed when having to turn a tablet 180 degrees and
this should never come up.
=gerry=

bobjacobsen, 2017-06-30:
If the code is mixing Swing orientation (which is about reading order: Hebrew is read from right to left, etc) and rotation (which is not), then that's something to resolve in any case.

geowar1, 2017-09-22:
I think that this issue was resolved in PR https://github.com/JMRI/JMRI/pull/3800. If wolfsonga would confirm we could close this one.

wolfsonga, 2017-09-22:
Nop, chopping is still happening. I am running 4.9.4 and any time I add
long strings of text, I have to pad it with 5 or more spaces. If my string
ends with => as a pointer, the text gets cut off at just before the last
text character preceeding it.
=gerry=

closed 2021-04-14 by stale[bot]