JMRI/JMRI #3602: Decoder Pro won't start on Windows 10

jakuta1, 2017-06-04:
I had Decoder Pro 4.4 working on the same laptop (64-bit) before upgrading to 4.6.  I've tried re-installing multiple times without success and even tried re-installing 4.4.  That no longer works either.  I'm not even getting a splash screen.  I've verified that Java 1.8.0_131 is installed by using java -version in a command window.  I'm using an NCE USB Interface and the latest drivers have been installed for that.  I've also tried forcing Decoder Pro to start in 32 bit mode, but that doesn't fix the issue.  The debug output is as follows:

CommandLine: "C:\Program Files (x86)\JMRI\LaunchJMRI.exe" /debug apps.gui3.dp3.DecoderPro3
Default options: 
AppName: DecoderPro3
Class: apps.gui3.dp3.DecoderPro3
Parameters: 
Noisy: 6
Force32bit: 0
Profile: 
Testing for x64...
Result: 1
Setting x64 registry view...
Setting x86 registry view...
JavaPath: C:\Program Files (x86)\Java\jre1.8.0_131\bin\javaw.exe
Delete file: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Remove folder: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\
Create folder: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI
CopyFiles: C:\Program Files (x86)\Java\jre1.8.0_131\bin\javaw.exe C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Result: 0
JExePath: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
PhysicalMemory: 3981m
Check if more than 4GB memory
Less than 4GB
Check if more than 1GB memory
More than 1GB
InitHeap: 96m
MinMemory: 192m
MaxMemory: 1990m
Checking maximum heap size...
Running x86 JVM
Adjusted MaxMemory: 1024m
...finished
Checking user.home...
Reading Desktop Shell Folder registry key...
...read: C:\Users\Jeffrey\Desktop
Checking if equal to %USERPROFILE%...
Comparing: C:\Users\Jeffrey
to: C:\Users\Jeffrey
user.home OK
Options:   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1024m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\Jeffrey\JMRI\systemfiles" -Djmri.log.path="C:\Users\Jeffrey\JMRI\systemfiles\\"
ClassPath: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.0.6.jar;lib\jackson-core-2.0.6.jar;lib\jackson-databind-2.0.6.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython-standalone-2.7.0.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\WinRegistry-4.4.jar;lib\xAPlib.jar;lib\xbjlib-1.1.1.jar;lib\xercesImpl.jar
Check for any prepended/appended classpath entries
Final ClassPath: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.0.6.jar;lib\jackson-core-2.0.6.jar;lib\jackson-databind-2.0.6.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython-standalone-2.7.0.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\WinRegistry-4.4.jar;lib\xAPlib.jar;lib\xbjlib-1.1.1.jar;lib\xercesImpl.jar
MaxLen: 8192
ExeString: "C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe"   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1024m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\Jeffrey\JMRI\systemfiles" -Djmri.log.path="C:\Users\Jeffrey\JMRI\systemfiles\\" -Djava.class.path=".;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.0.6.jar;lib\jackson-core-2.0.6.jar;lib\jackson-databind-2.0.6.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython-standalone-2.7.0.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\WinRegistry-4.4.jar;lib\xAPlib.jar;lib\xbjlib-1.1.1.jar;lib\xercesImpl.jar" apps.gui3.dp3.DecoderPro3 
Output folder: C:\Program Files (x86)\JMRI
Launching JMRI
Execute: "C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe"   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1024m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\Jeffrey\JMRI\systemfiles" -Djmri.log.path="C:\Users\Jeffrey\JMRI\systemfiles\\" -Djava.class.path=".;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.0.6.jar;lib\jackson-core-2.0.6.jar;lib\jackson-databind-2.0.6.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython-standalone-2.7.0.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio [... cut ...] -1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\WinRegistry-4.4.jar;lib\xAPlib.jar;lib\xbjlib-1.1.1.jar;lib\xercesImpl.jar" apps.gui3.dp3.DecoderPro3 
Return code from process: 1
To copy this text to the clipboard, right click then choose
  'Copy Details To Clipboard'
Completed

Any help would be much appreciated.

Thanks,

Jeff

bobjacobsen, 2017-06-04:
Please try the current development build at http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

It has several updates to the Windows installer.

jakuta1, 2017-06-04:
> Please try the current development build at http://jmri.tagadab.com/jenkins/job/Development/job/Packages/It has several updates to the Windows installer.—

Hi Bob,
I installed the current development build and ran into the same issue.  Here is the debug output from the latest run. I think it is basically the same as before:
CommandLine: "C:\Program Files (x86)\JMRI\LaunchJMRI.exe" /debug apps.gui3.dp3.DecoderPro3
Default options: 
AppName: DecoderPro3
Class: apps.gui3.dp3.DecoderPro3
Parameters: 
Noisy: 6
Force32bit: 0
Profile: 
Testing for x64...
Result: 1
Setting x64 registry view...
Setting x86 registry view...
JavaPath: C:\Program Files (x86)\Java\jre1.8.0_131\bin\javaw.exe
Delete file: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Remove folder: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\
Create folder: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI
CopyFiles: C:\Program Files (x86)\Java\jre1.8.0_131\bin\javaw.exe C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Result: 0
JExePath: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
PhysicalMemory: 3981m
Check if more than 4GB memory
Less than 4GB
Check if more than 1GB memory
More than 1GB
InitHeap: 96m
MinMemory: 192m
MaxMemory: 1990m
Checking maximum heap size...
Running x86 JVM
Adjusted MaxMemory: 1024m
...finished
Checking user.home...
Reading Desktop Shell Folder registry key...
...read: C:\Users\Jeffrey\Desktop
Checking if equal to %USERPROFILE%...
Comparing: C:\Users\Jeffrey
to: C:\Users\Jeffrey
user.home OK
Options:   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1024m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\Jeffrey\JMRI\systemfiles" -Djmri.log.path="C:\Users\Jeffrey\JMRI\systemfiles\\"
ClassPath: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.8.5.jar;lib\jackson-core-2.8.5.jar;lib\jackson-databind-2.8.5.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\jna-4.4.0.jar;lib\jna-platform-4.4.0.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython-standalone-2.7.0.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar
Check for any prepended/appended classpath entries
Final ClassPath: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.8.5.jar;lib\jackson-core-2.8.5.jar;lib\jackson-databind-2.8.5.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\jna-4.4.0.jar;lib\jna-platform-4.4.0.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython-standalone-2.7.0.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar
MaxLen: 8192
ExeString: "C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe"   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1024m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\Jeffrey\JMRI\systemfiles" -Djmri.log.path="C:\Users\Jeffrey\JMRI\systemfiles\\" -Djava.class.path=".;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;li [... cut ...] \slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar" apps.gui3.dp3.DecoderPro3 
Return code from process: 1
To copy this text to the clipboard, right click then choose
  'Copy Details To Clipboard'
Completed

Regards,
Jeff

rhwood, 2017-06-05:
Can you please add the file C:\Users\Jeffrey\JMRI\systemfiles\session.log to this issue (I think this is the correct path)? You may have to change its extension to .txt or zip it to upload it.

jakuta1, 2017-06-05:
Hi Randall,
I would have been glad to upload the session.log file except that there isn't such a file in C:\Users\Jeffrey\JMRI\systemfiles.  Apparently it is not being created when I attempt to start Decoder Pro.

Regards,
Jeff

rhwood, 2017-06-05:
Look for one anywhere under the JMRI folder.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

jakuta1, 2017-06-05:
Randall,

I wasn't able to find that file anywhere.
Jeff

   

http://alexandriasoftware.com

rhwood, 2017-06-06:
Without that log, I am at a bit of a loss. Can you try adding the "/noisy" parameter to your launch shortcut (after "/debug") to see if that presents some additional logging?

Can you also see if your computer's malware protection systems are blocking any applications from running?

jakuta1, 2017-06-10:
Hi Randall,
Sorry for the delay in getting back to you.  Below is the debug output when the "/noisy" parameter is added:
CommandLine: "C:\Program Files (x86)\JMRI\LaunchJMRI.exe" /debug /noisy apps.gui3.dp3.DecoderPro3
Default options: 
AppName: DecoderPro3
Class: apps.gui3.dp3.DecoderPro3
Parameters: 
Noisy: 1
Force32bit: 0
Profile: 
Testing for x64...
Result: 1
Setting x64 registry view...
Setting x86 registry view...
JavaPath: C:\Program Files (x86)\Java\jre1.8.0_131\bin\java.exe
Delete file: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Remove folder: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\
Create folder: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI
CopyFiles: C:\Program Files (x86)\Java\jre1.8.0_131\bin\java.exe C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Result: 0
JExePath: C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
PhysicalMemory: 3981m
Check if more than 4GB memory
Less than 4GB
Check if more than 1GB memory
More than 1GB
InitHeap: 96m
MinMemory: 192m
MaxMemory: 1990m
Checking maximum heap size...
Running x86 JVM
Adjusted MaxMemory: 1024m
...finished
Checking user.home...
Reading Desktop Shell Folder registry key...
...read: C:\Users\Jeffrey\Desktop
Checking if equal to %USERPROFILE%...
Comparing: C:\Users\Jeffrey
to: C:\Users\Jeffrey
user.home OK
Options:   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1024m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\Jeffrey\JMRI\systemfiles" -Djmri.log.path="C:\Users\Jeffrey\JMRI\systemfiles\\"
ClassPath: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.8.5.jar;lib\jackson-core-2.8.5.jar;lib\jackson-databind-2.8.5.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\jna-4.4.0.jar;lib\jna-platform-4.4.0.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython-standalone-2.7.0.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar
Check for any prepended/appended classpath entries
Final ClassPath: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.8.5.jar;lib\jackson-core-2.8.5.jar;lib\jackson-databind-2.8.5.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.3.9.v20160517.jar;lib\jetty-io-9.3.9.v20160517.jar;lib\jetty-security-9.3.9.v20160517.jar;lib\jetty-server-9.3.9.v20160517.jar;lib\jetty-servlet-9.3.9.v20160517.jar;lib\jetty-util-9.3.9.v20160517.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns.jar;lib\jna-4.4.0.jar;lib\jna-platform-4.4.0.jar;lib\joal.jar;lib\jul-to-slf4j-1.7.13.jar;lib\jython-standalone-2.7.0.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\RXTXcomm.jar;lib\Serialio.jar;lib\slf4j-api-1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar
MaxLen: 8192
ExeString: "C:\Users\Jeffrey\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe"   -noverify -Dsun.java2d.d3d=false -Djava.security.policy=security.policy -Djinput.plugins=net.bobis.jinput.hidraw.HidRawEnvironmentPlugin -Dswing.defaultlaf=com.sun.java.swing.plaf.windows.WindowsLookAndFeel -Djava.library.path=.;lib\x86;lib -Djava.rmi.server.codebase=file:java/classes/ -Dsun.java2d.noddraw -Xms96m -Xmx1024m -Dfile.encoding=UTF-8 -Dpython.home="C:\Users\Jeffrey\JMRI\systemfiles" -Djmri.log.path="C:\Users\Jeffrey\JMRI\systemfiles\\" -Djava.class.path=".;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\ch.ntb.usb.jar;lib\commons-io-1.3.1.jar;lib\commons-lang3-3.2.1.jar;lib\gluegen-rt.jar;lib\jackson-annotations-2.8 [... cut ...] -1.7.13.jar;lib\slf4j-log4j12-1.7.13.jar;lib\smtp.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.3.9.v20160517.jar;lib\websocket-client-9.3.9.v20160517.jar;lib\websocket-common-9.3.9.v20160517.jar;lib\websocket-server-9.3.9.v20160517.jar;lib\websocket-servlet-9.3.9.v20160517.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xercesImpl.jar" apps.gui3.dp3.DecoderPro3 
Return code from process: 1
To copy this text to the clipboard, right click then choose
  'Copy Details To Clipboard'
Completed

jakuta1, 2017-06-25:
Randall, I wanted to let you know that I was finally able to get Decoder Pro 4.7.6.R5 to startup.  You were correct that it had to do with the anti-virus software I was using (called Comodo).  Unfortunately, the solution required uninstalling Comodo since I was unable to unblock the problem it was causing with JMRI using Comodo's user interface.
Thanks,
Jeff

rhwood, 2017-06-25:
@jakuta1 That is good to know. Unfortunately I can't help with your AV software, but maybe this will? https://help.comodo.com/topic-72-1-451-4758-.html

closed 2017-06-25 by rhwood