JMRI/JMRI #3608: Block Tracking problem with linked LE panels

dsand47, 2017-06-05:
Block tracking does not work properly when two layout editor panels are linked using edge connectors.  This problem was introduced in the 4.7.1 or 4.7.2.

The attached file contains 2 test panel xml files and the session logs.  With the unlinked version, the block tracking works as expected within each panel.  For the linked version, the block tracking has errors on the left panel, but not the right.  At the edge boundary, going from left to right tracking starts working.  When going right to left, it stops working at the edge boundary.

A review of the XML file for the test case and a real case indicates that the first LE panel in the XML file has the problem.  This was confirmed by flipping the order of the LE panels in the XML file.

[Edit Link Test.zip](https://github.com/JMRI/JMRI/files/1052339/Edit.Link.Test.zip)

Dave Sand

dsand47, 2017-06-10:
I have created PR #3623.  This PR has a proposed solution to the block tracking problem when edge connectors are used.

It prevents duplicate connection entries in the **cList** array which then eliminates the duplicate entries in the **paths** array which causes the tracking problem.

Due to the complexity of the code feeding **LayoutEditorAuxTools.checkConnectivity()**, this seemed like the best solution.

Simple and complex layout editor panels tested ok.  A local run of JMRI tests had no errors.

I tried the Git commands shown in issue #3619 but they did not work as expected so the PR will need manual intervention in order to be included in 4.7.6/4.8.

Dave Sand

geowar1, 2017-08-13:
Has this been resolved? Can this issue be closed?

dsand47, 2017-08-17:
Tested 4.8 and block tracking works as expected.

closed 2017-08-17 by dsand47