JMRI/JMRI #3616: Rotated text in Layout Editor doesn't look good

bobjacobsen, 2017-06-07:
From Gerald Wolfson "wolfsonga@gmail.com" via email:

> I have included an attach jpeg which shows the results at 45, 90, 135, and 180. I believe there is a major flaw in the text rotation code inside of JMRI but this is not my area of expertise. Could you pass this on to someone who might be able to fix it. By the way, the last two text strings on the right were created with Orientation not Rotation.

[image]

bobjacobsen, 2017-06-07:
More from Gerald Wolfson:

> Here is the .xml I used to generate and similar image to the jpeg I sent you.

[TextRotation.xml.txt](https://github.com/JMRI/JMRI/files/1058901/TextRotation.xml.txt)

geowar1, 2017-06-26:
See Pull Request https://github.com/JMRI/JMRI/pull/3684

geowar1, 2017-06-26:
JRML admins: Please assign. TIA.

geowar1, 2017-07-20:
Ok… been playing with this code… ouch… Just for giggles set the rotation to any non-zero value… then set it back to zero… NullPointerException (NPE)! 1st bug: when you try to reset rotation to zero the code attempts to restore original icon from its (file) url… but since this icon was created (in memory) just to hold the rotated text it doesn't have a url. So… NPE…

rhwood, 2017-07-20:
You shouldn't need an icon to rotate text. See https://stackoverflow.com/questions/10083913/how-to-rotate-text-with-graphics2d-in-java for examples.

geowar1, 2017-07-21:
PositionableLabel's can show text or an icon or both… the way the code is written if the label is rotated then the icon & text are rendered (rotated) into a BufferedImage and cached (at setRotation() time). I assume this is so the rotation doesn't have to happen at draw time? (shouldn't matter but… ;-)

geowar1, 2017-07-21:
Finding more bugs… if you change the rotation to a non-zero (1° for example) then you can't change the font size or style unless you set the rotation back to zero first (the rotated text won't re-render with the new font size or style). I could fix this except I think I'd rather just pull all the code that renders rotated text into a (cached) buffer (and drawing it rotated at paint() time instead)…

rhwood, 2017-07-21:
Can you do https://stackoverflow.com/a/26006540/176160 <https://stackoverflow.com/a/26006540/176160> if you cache the rendered text as a buffered image? I understand our current code can’t rotate an animated gif...

Randall Wood
JMRI Developer - http://jmri.org
Alexandria Software - http://alexandriasoftware.com
Northern Virginia NTRAK - http://nvntrak.org

geowar1, 2017-07-21:
On Jul 21, 2017, at 3:13 PM, Randall Wood <notifications@github.com> wrote:
> On Jul 20, 2017, at 23:00, George Warner <notifications@github.com> wrote: 
>> PositionableLabel's can show text or an icon or both… the way the code is written if the label is rotated then the icon & text are rendered (rotated) into a BufferedImage and cached (at setRotation() time). I assume this is so the rotation doesn't have to happen at draw time? (shouldn't matter but… ;-)
> Can you do https://stackoverflow.com/a/26006540/176160 <https://stackoverflow.com/a/26006540/176160> if you cache the rendered text as a buffered image? I understand our current code can’t rotate an animated gif…

First let me correct myself… if the label is rotated then ether the text or the text and icon (if an icon is defined) are rendered (non-rotated) into a NamedIcon… at paint time that icon is drawn after a                  g2d.transform(AffineTransform.getRotateInstance(Math.toRadians(_degrees)));

Exactly as the stack overflow example does it.

The problem appears to be that rendering the text into the buffered image is what causes the blurring problem.

My current plan of attack is to skip the step of compositing the text with the icon and just draw the text after I’ve transform’d the g2d with getRotateInstance(Math.toRadians(_degrees)));.

-- 
Enjoy,
George Warner
eMail/iChat: <geowar1@mac.com>

closed 2021-04-14 by stale[bot]