JMRI/JMRI #3619: Create test release 4.7.6

bobjacobsen, 2017-06-09:
JMRI test release 4.7.6 is planned to be made from the release-4.7.5 branch plus minor additions and fixes. It is tentatively scheduled for build on June 15, 2017, a week from now. This is intended to be the first, and hopefully the last, clean-up test release before production release 4.8 in late June or early July.

If you think a particular PR should be included, please add a note here.

If you're going to create something that you'd like to have hear, please create your branch from the release-4.7.5 branch:
```
git fetch origin
git checkout origin/release-4.7.5
git branch -b my-new-branch-name
```

That can be easily merged via PR onto both `master` (for the main line of development) and release 4.7.6 and any successors.

bobjacobsen, 2017-06-10:
#3608 infers that PR #3623 "Prevent duplicate connections" should be included in JMRI 4.7.6

bobjacobsen, 2017-06-11:
Added #3623 to release-4.7.6 branch
```
% git cherry-pick 6768a48891fcfe5c4b5d42fdd0ab5cba01b4696d
[release-4.7.6 e0bef595fa] Prevent duplicate connections
 Author: dsand47 <gh@dfm.sandsys.org>
 Date: Sat Jun 10 11:44:20 2017 -0500
 1 file changed, 19 insertions(+), 1 deletion(-)
```

bobjacobsen, 2017-06-11:
Added #3624 to release-4.7.6 branch:
```
% git cherry-pick bd564ece6d817aff9bba38f66e467fd5d0cf3456 71c71bc600592c49f6de48efe7742b1f2807918a
[release-4.7.6 f47975044a] Added customs scale selection, at request of Bachrus
 Author: crosland <andrewcrosland@talktalk.net>
 Date: Sat Jun 10 19:18:03 2017 +0100
 5 files changed, 63 insertions(+), 18 deletions(-)
[release-4.7.6 201cd9f06f] Correct error due to bads edit of .properties
 Author: crosland <andrewcrosland@talktalk.net>
 Date: Sat Jun 10 21:00:06 2017 +0100
 1 file changed, 1 insertion(+), 1 deletion(-)
```

KenC57, 2017-06-13:
#3631 is currently being tested. Since it was only a simple decoder fix, it should pass shortly. When it does, it should be added to 4.7.6.

bobjacobsen, 2017-06-13:
Add #3628 to release-4.7.6 branch:
```
% git cherry-pick 7ada2ebc286836e73a459a20f6bf5557496112a4
[release-4.7.6 41c3835020] Danish translation
 Author: SonnysHansen <sonnyshansen@gmail.com>
 Date: Sun Jun 11 16:52:44 2017 +0200
 1 file changed, 41 insertions(+), 42 deletions(-)
```

bobjacobsen, 2017-06-13:
Added #3631 to release-4.7.6 branch:
```
% git cherry-pick f25e5301f3ec437b2409eddcf2f7fb22b3a64133
[release-4.7.6 3bad8fe926] Update to TD_SIC24AD decoder from Team Digital.
 Author: Ken Cameron <kcameron@twcny.rr.com>
 Date: Mon Jun 12 20:19:30 2017 -0400
 1 file changed, 2 insertions(+), 1 deletion(-)
```

bobjacobsen, 2017-06-14:
Added #3637 to the release-4.7.6 branch:
```
% git cherry-pick 1679c2c1259ed9abcf305e744f5e1d3aef36f982
[release-4.7.6 92d9d17d15] Do not exit when exiting
 Author: Randall Wood <randall.h.wood@alexandriasoftware.com>
 Date: Wed Jun 14 06:20:17 2017 -0400
 1 file changed, 3 insertions(+)
```

bobjacobsen, 2017-06-14:
#3635 merged in the usual way

bobjacobsen, 2017-06-17:
JMRI 4.7.6 has been released.

closed 2017-06-17 by bobjacobsen