JMRI/JMRI #3620: Cancel profile selector dialog leave program running with no windows

bobjacobsen, 2017-06-09:
From @devel-bobm on jmri-developers thread about building 4.7.5:

> Win7-64.  Start PanelPro for a user with many configuration profiles.
Window comes up for selecting the configuration profile.  If I close that
window using the "X" in the upper right corner, the selector disappears, but
the "Launcher" is not closed - Task Manager shows both "LaunchJMRI.exe" and
"PanelPro.exe" as alive.  If I kill the "PanelPro.exe" process, it will
automatically stop the "LaunchJMRI.exe" process.
> 
> DecoderPro for this release candidate do not exhibit this behavior - the "X"
properly closes DecoderPro.exe as well as LaunchJMRI.exe.
> 
> I have a few older releases installed.  PanelPro 4.3.1 and 4.3.2 do not
exhibit this behavior.  PanelPro 4.5.3 does.

devel-bobm, 2017-06-13:
It appears that if the `DefaultShutdownManager` has been installed before `ProfileManagerDialog.getStartingProfile` is invoked, then "X"ing the dialog box will not shut down the program.  PanelPro installs the DefaultShutdownManager before presenting the dialog box for the configuration profile; DecoderPro3 installs the DefaultShutdownManager after presenting the dialog box for the configuration profile.

DecoderPro3 inherits from the constructor `jmri.apps.AppsBase(String applicationName, String configFileDef, String[] args)`, which includes:
`    public AppsBase(String applicationName, String configFileDef, String[] args) {

        if (!preInit) {
            preInit(applicationName);
            setConfigFilename(configFileDef, args);
        }

        Log4JUtil.initLogging();

            configureProfile();

        installConfigurationManager();

        installShutDownManager();

        addDefaultShutDownTasks();
`

PanelPro inherits from jmri.apps.Apps, with a constructor that does a lot of various stuff, before installing the DefaultShutdownManager and then dealing with the profile selection dialog box:
...
`        // install shutdown manager
        InstanceManager.setDefault(ShutDownManager.class, new DefaultShutDownManager());

        // add the default shutdown task to save blocks
        // as a special case, register a ShutDownTask to write out blocks
        InstanceManager.getDefault(jmri.ShutDownManager.class).
                register(new AbstractShutDownTask("Writing Blocks") {
                    @Override
                    public boolean execute() {
'
...
`        // @see jmri.profile.ProfileManager#migrateToProfiles Javadoc for conditions handled here
        if (!ProfileManager.getDefault().getConfigFile().exists()) { // no profile config for this app
            try {
                if (ProfileManager.getDefault().migrateToProfiles(configFilename)) { // migration or first use
                    // notify user of change only if migration occured
                    // TODO: a real migration message
                    JOptionPane.showMessageDialog(sp,
                            Bundle.getMessage("ConfigMigratedToProfile"),
                            jmri.Application.getApplicationName(),
                            JOptionPane.INFORMATION_MESSAGE);
                }
            } catch (IOException | IllegalArgumentException ex) {
`
...

If, in jmri.apps.Apps,  I move the installation of DefaultShutdownManager to be after the configuration profile dialog, the "X" successfully cancels PanelPro, solving the problem.

Is this a reasonable approach?

devel-bobm, 2017-06-13:
For example, `devel-bobm/JMRI`, branch `dbm-panelpro_startup` [https://github.com/devel-bobm/JMRI/tree/dbm-panelpro_startup](url)

rhwood, 2017-06-13:
If you just move the call to installDefaultShutdownTasks to after selecting the profile (leaving the ShutdownManager defined before the Profile is selected) does that work?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

devel-bobm, 2017-06-14:
@rhwood, I did not find "installDefaultShutdownTasks".  I find that `addDefaultShutdownTasks` is used in AppBase.java, and it is invoked _after_ `configureProfile()`.  It is invoked as part of DecoderPro3 start-up, but is not invoked as part of PanelPro start-up.

I do not find invocation of `addDefaultShutdownTasks` in the start-up of PanelPro.  It does not execute the code in AppBase.java; it inherits from Apps.java, not AppBase.java.  So, instead, I moved some code which performs registration of some shutdown tasks from before the configuration profile dialog box invocation to after the code for the dialog box.  I moved this code:
`        try {
            ProfileManagerDialog.getStartingProfile(sp);
            // Manually setting the configFilename property since calling
            // Apps.setConfigFilename() does not reset the system property
            configFilename = FileUtil.getProfilePath() + Profile.CONFIG_FILENAME;
            System.setProperty("org.jmri.Apps.configFilename", Profile.CONFIG_FILENAME);
            log.info("Starting with profile {}", ProfileManager.getDefault().getActiveProfile().getId());
        } catch (IOException ex) {
            log.info("Profiles not configurable. Using fallback per-application configuration. Error: {}", ex.getMessage());
        }`

This caused the hang, so no improvement over current JMRI/JMRI "head".

rhwood, 2017-06-14:
Calling System.exit() after System.exit() has already been called blocks indefinitely (see https://docs.oracle.com/javase/7/docs/api/java/lang/Runtime.html#exit(int)) (this is hinted at in a comment in the DefaultShutDownManager)

Since the DefaultShutDownManager attempts to close all non-disposed windows (this is required to allow Windows to prompt to save changes when using the Application->Quit menu item in macOS), and the ProfileManagerDialog calls exit directly when closing, this causes the application to never exit.

Therefor the correct action to fix this is to modify the ProfileManagerDialog so that it does not call exit if exit has already been called.

rhwood, 2017-06-14:
This has been fixed for 4.9.1 with 47b34b9, but needs to be fixed for 4.7.6 as well.

bobjacobsen, 2017-06-14:
Added to JMRI 4.7.6

devel-bobm, 2017-06-14:
Commit 47b34b9 does not solve the original problem.

rhwood, 2017-06-14:
What threads does a debugger show are open after this fails to exit?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

devel-bobm, 2017-06-14:
See attached image, from PanelPro debug session under NetBeans on Win7-64:
[image]

devel-bobm, 2017-06-14:
FWIW, I can use the debugger to "Interrupt" any of those threads.  Only "interrupting" the "AWT-EventQueue-0" thread causes PanelPro to terminate.

devel-bobm, 2017-06-14:
Another piece that could be of some use - the threads **which are alive while the dialog box is open**:

[image]

rhwood, 2017-06-15:
It'll be until this weekend until I can make the ProfileManagerDialog behave correctly (exit the app _after_ the window is closed and disposed of, not while closing it, so that code that attempts to close all open windows does not trip up on it). Since this was in 4.6, I think there is no need to hold 4.7.6 on this.

rhwood, 2017-06-15:
I have been unable to recreate this problem on macOS, even in a new build VM, so I think it is in some way unique to how Windows handles windows.

rhwood, 2017-06-15:
@devel-bobm Please test my [master branch](https://github.com/rhwood/jmri/tree/master), it may address the issue. If not, I need a running stack on the hang, as this appears to be an issue unique to Microsoft Windows.

devel-bobm, 2017-06-15:
Out of town - hope to try tonite (US East coast).

devel-bobm, 2017-06-16:
@rhwood, your master exhibits the same failure as originally reported.

Code condition at time of testing:
`git pull` (from JMRI/JMRI master) resulted in 
'...From https://github.com/JMRI/JMRI
   47b34b9..34e783c  master        -> origin/master...'
then `git checkout -b newBranchName`
then `git pull https://github.com/rhwood/JMRI.git master`

If I then try moving the `DefaultShutdownManager` installation and 'DefaultShutdownTask" registration to _after_ the configuration profile dialog, like I did before, now instead of getting a clean shutdown, I get an unhandled exception which terminates successfully:
`2017-06-15 22:22:54,306 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.lang.IllegalStateException: Shutdown in progress
	at java.lang.ApplicationShutdownHooks.add(ApplicationShutdownHooks.java:66)
	at java.lang.Runtime.addShutdownHook(Runtime.java:211)
	at jmri.managers.DefaultShutDownManager.<init>(DefaultShutDownManager.java:61)
	at apps.Apps.<init>(Apps.java:188)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)
	at apps.PanelPro.PanelPro.main(PanelPro.java:104)
Java Result: 255
apps.PanelPro.PanelPro finished with return code 255
Execute failed: java.io.IOException: Cannot run program "git" (in directory "E:\User Data\NetBeans\JMRI"): CreateProcess error=2, The system cannot find the file specified
BUILD SUCCESSFUL (total time: 1 minute 7 seconds)`

I have not seen the above exception before - only with your branch's "master" plus a move of the installed `ShutdownManager` plus move of the registered `DefaultShutdownTask`.

rhwood, 2017-06-16:
On macOS, I was able to create a similar (if not the same) failure as originally reported, by closing the JMRI application with Cmd-Q (but not by clicking the close window button). In that case, the DefaultShutDownManager deadlocked on a synchronized wait while trying to close a window identified as "Frame0" with title "JMRI". I think that window is the "Press F8 to disable Logix" window, but have been unable to confirm as of yet.

#3642 fixes this by ensuring that the ProfileManagerDialog's call to ```System.exit()``` is before the default instance of the ShutDownManager is created (although that instance does get created because the App constructor continues and creates it before it gets killed while exiting).

commit 47b34b9a (2017-06-14): Merge pull request #3637 from rhwood/master

Prevent hang exiting PanelPro at Profile selector.
Fixes #3620 for 4.9.1
commit b9224cb0 (2017-06-17): Merge pull request #3642 from rhwood/master

Fix issues with Profile selection
Fixes #3620 
Fixes #3641 
Fixes #3639 
Fixes #3452
closed 2017-06-17 by rhwood