JMRI/JMRI #3622: NPE from scroll wheel operation

bobjacobsen, 2017-06-10:
From a jmriusers conversation:

> Just noticed this, I normally don't use the scroll wheel on the mouse in the layout, but I did this time and got the below. Doesn't happen in edit mode, that scrolls as expected. (Fedora F25 x86_64/gnome/digitrax pr3) 

```
2017-06-09 20:54:52,811 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: null source
at java.util.EventObject.<init>(EventObject.java:56)
at java.awt.AWTEvent.<init>(AWTEvent.java:337)
at java.awt.event.ComponentEvent.<init>(ComponentEvent.java:120)
at java.awt.event.InputEvent.<init>(InputEvent.java:341)
at java.awt.event.MouseEvent.<init>(MouseEvent.java:736)
at java.awt.event.MouseWheelEvent.<init>(MouseWheelEvent.java:302)
at java.awt.event.MouseWheelEvent.<init>(MouseWheelEvent.java:248)
at jmri.jmrit.display.layoutEditor.LayoutEditor.mouseWheelMoved(LayoutEditor.java:3347)
at java.awt.Component.processMouseWheelEvent(Component.java:6621)
at java.awt.Component.processEvent(Component.java:6305)
at java.awt.Container.processEvent(Container.java:2236)
at java.awt.Component.dispatchEventImpl(Component.java:4889)
at java.awt.Container.dispatchEventToSelf(Container.java:2318)
at java.awt.Component.dispatchMouseWheelToAncestor(Component.java:5042)
at java.awt.Component.dispatchEventImpl(Component.java:4776)
at java.awt.Container.dispatchEventImpl(Container.java:2294)
at java.awt.Component.dispatchEvent(Component.java:4711)
at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4554)
at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
at java.awt.Container.dispatchEventImpl(Container.java:2280)
at java.awt.Window.dispatchEventImpl(Window.java:2746)
at java.awt.Component.dispatchEvent(Component.java:4711)
at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
at java.awt.EventQueue.access$500(EventQueue.java:97)
at java.awt.EventQueue$3.run(EventQueue.java:709)
at java.awt.EventQueue$3.run(EventQueue.java:703)
at java.security.AccessController.doPrivileged(Native Method)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
at java.awt.EventQueue$4.run(EventQueue.java:731)
at java.awt.EventQueue$4.run(EventQueue.java:729)
at java.security.AccessController.doPrivileged(Native Method)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

geowar1, 2017-06-10:
This is probably mine… please assign. I'll take a look next week when I'm back home.

geowar1, 2017-10-29:
I think that this has been fixed… Sorry, don't remember the PR. Can someone (OP?) please regress and close if fixed?

closed 2017-10-29 by bobjacobsen