JMRI/JMRI #3626: Extraneous -psn files created when activating a Profile under OSX.

CCatania01, 2017-06-10:
When running 4.7.5 PanelPro, activating a Profile creates a file named -psn_n_nnn in the ~/Library/JMRI directory.  These files contain profile parameters.  Each profile activation creates a new file which persists.

This is the contents of a -psn file
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE properties SYSTEM "http://java.sun.com/dtd/properties.dtd">
<properties>
<comment>Active profile configuration (saved at Sat Jun 10 07:14:55 PDT 2017)</comment>
<entry key="autoStart">false</entry>
<entry key="autoStartTimeout">10</entry>
<entry key="activeProfile">Profile_John_Pearson.3eecd5f2</entry>
</properties>

I request this feature be removed as the files clutter up the JMRI User directory.

rhwood, 2017-06-11:
This is an issue only if the user does not follow the installation guidance at http://jmri.org/install/MacOSX.shtml

CCatania01, 2017-06-18:
4.6-R81496dc has the same behavior.  I properly installed that production version from the JMRI site less than an hour ago.

rhwood, 2017-06-23:
The macOS ```open``` command (and probably Gatekeeper in some situations) adds a parameter to all programs that allows the program to know its ProcessSerialNumber. Cocoa and Carbon (i.e. native macOS programs handle this automatically, but others do not necessarily do so).

This is fixed for JMRI 4.9.1 and later releases.

commit 50f09b25 (2017-06-23): Merge pull request #3671 from JMRI/rhwood-patch-2

Ignore the ProcessSerialNumber argument
Fixes #3626 in JMRI 4.9.1
closed 2017-06-23 by rhwood