JMRI/JMRI #3627: VersionEye CI step consistently failing

bobjacobsen, 2017-06-11:
The VersionEye CI step has started to fail every CI cycle. It's flagging several of the libraries we include as having "UKNOWN" licenses.

[Typical fail report](https://www.versioneye.com/pullrequests/593c40160fb24f0077203e6b)

[image]

rhwood, 2017-06-11:
I don't know why and will remove it Monday.

bobjacobsen, 2017-06-13:
#3629 showed an interestingly different case:  VersionEye flagged SerialIO as not having a license.  But SerialIO shouldn't be present any more....

rhwood, 2017-06-13:
I think that VersionEye is checking the dependencies in the branch to be merged, not the results of merging the PR, so in #3629, where @marcfour has not updated his master from JMRI's master, SerialIO is still shown as a dependency.

bobjacobsen, 2017-06-13:
@rhwood Thanks!  Hadn't thought of that, and it definitely explains it.

marcfour, 2017-06-13:
Question on this. Looked high and low.. How do I update my master from JMRI's master short of delete and re fork.

bobjacobsen, 2017-06-14:
> How do I update my master from JMRI's master short of delete and re fork.
@marcfour: A bit here, but if this goes on we should move it to the jmri-developers list.

Background here:  http://jmri.org/help/en/html/doc/Technical/gitdeveloper.shtml  with an introduction at http://jmri.org/help/en/html/doc/Technical/getgitcode.shtml  

Once you've got the connection to GitHub set up, if you're using git from the command line, you should just have to do:
```
git pull origin master
```

That says to go to the `origin` repository, which should be already configured to point to GitHub, and then merge the contents of the `master` branch with the current content in your directory.

rhwood, 2017-06-23:
Closing as this is no longer happening.

closed 2017-06-23 by rhwood