JMRI/JMRI #3633: Possible Improvements to MultiDecoder Tool

bobjacobsen, 2017-06-13:
In a jmri-developers conversation, @dheap listed some complexities is S9.2.2 Decoder Locking that the [Multi-Decoder Tool](http://jmri.org/help/en/html/apps/DecoderPro/MultiDecoder.shtml) doesn't handle well:

- [ ] ESU decoders accept a range of 0-255 as valid Lock IDs. SoundTraxx decoders accept a range of 0-7 as valid Lock IDs. (S9.2.2 doesn't specify a range.)
- [ ] TCS decoders accept a value of 0 in CV15 as a 'universal unlock key'. From the TCS documentation "To unlock a decoder, make CV 15 = 0 or CV 15 = CV 16." (S9.2.2 doesn't suggest that possibility.) I've been bitten by that one in a real-world situation.
- [ ] Some manufacturers have a separate "CV15/16 Lock Enable" setting. For example ESU uses CV124 bit 1 and SoundTraxx uses CV30 bit 0. If you don't set the Lock Enable bit, the decoder is unlocked irrespective of CV15/CV16 values (S9.2.2 doesn't suggest that possibility.)

The code and/or documentation could be updated to help people with these. (A separate conversation is ongoing about adding support to DecoderPro itself in various ways)

dheap, 2017-06-13:
Will there still be a need for this tool as a separate entity if/when we add a new variable type to DecoderPro definitions, where context-specific idiosyncrasies can be more easily catered for?

bobjacobsen, 2017-06-14:
> Will there still be a need for this tool as a separate entity if/when we add a new variable type to DecoderPro definitions, where context-specific idiosyncrasies can be more easily catered for?

Hard to say.  It's certainly not all that popular now, though a few people do use it.  Others just use the Single CV programmer.  

Perhaps there are more "idiosyncrasies" than the list above, but those should be easily handled. If it worked well, I suspect a few people would use it.

closed 2021-04-14 by stale[bot]