JMRI/JMRI #3639: Empty profile key

silverailscolo, 2017-06-14:
@rhwood Key ``ProfileManagerDialog.profiles.AccessibleContext.accessibleName`` in [profile/Bundle.properties](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/profile/Bundle.properties) has no value even in the default language (used in line 140 of [ProfileManagerDialog.java](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/profile/ProfileManagerDialog.java)).
Is this intentional?

rhwood, 2017-06-15:
It's not intentional that it is empty, but I don't know that it matters, since this appears to be one of the few (if not the only) places we set an accessible name for an item. It should be "a localized String that designates the purpose of the object" (per the [Javadoc for AccessibleContext](http://docs.oracle.com/javase/6/docs/api/javax/accessibility/AccessibleContext.html#getAccessibleName())).

rhwood, 2017-06-16:
This remains a valid issue until it is actually fixed in JMRI/master.

commit b9224cb0 (2017-06-17): Merge pull request #3642 from rhwood/master

Fix issues with Profile selection
Fixes #3620 
Fixes #3641 
Fixes #3639 
Fixes #3452
closed 2017-06-17 by rhwood