JMRI/JMRI #3640: Script still use RXTX

rhwood, 2017-06-15:
Per a discussion on the Yahoo Users Group, some jython scripts still refer to gnu.io instead of purejavacomm.

> I have a problem with scripts that use import gnu.io for using USB ports. Scripts worked OK with JMRI 4.6. I installed 4.7.4 and then 4.7.5 and found that on both 4.7 versions I got a failure. System console reports - import error: no module named gnu.
> Using Windows 10 Java 1.8.
> These failures also happen with the example scripts such as - SerialPortTest.py . I am very much a cut and paste person so any help gratefully received.

(I can't log into Yahoo so I don't have a link.)

bobjacobsen, 2017-06-15:
The shipped-in-release example scripts were updated, but people's own versions (from scratch or copied-and-modified) have to be updated.  Will add that to the release notes.

rhwood, 2017-06-23:
There are scripts in master refering to gnu.io. See https://github.com/JMRI/JMRI/blob/master/jython/ComPortEnumeration.py

PR #3682 (2017-06-26): Replace gnu.io by purejavacomm in Python scripts
Closes #3640
closed 2017-06-26 by bobjacobsen