JMRI/JMRI #3641: Startup crash results in program unable to restart

bobjacobsen, 2017-06-15:
PanelPro took a JVM crash during an otherwise-normal startup, perhaps due to a missing USB device.

On the _next_ startup, a traceback window was presented; when OK clicked, the program terminated. See log below. This continues; repeating doesn't solve it. There's no user-level way to fix it by saving new preferences.  

- [ ] Ideally, the NPE would be protected against and the program could continue at least far enough to save new, valid preferences.

- [ ] Does PanelPro really have to write a critical file during a normal (no preferences changing) startup?  That seems to greatly raise the odds of corrupting preferences.

```
     [java] 2017-06-15 13:58:35,804 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
     [java] 2017-06-15 13:58:36,918 util.Log4JUtil                        INFO  - This log is appended to file: /Users/jake/Library/Preferences/JMRI/log/messages.log [main]
     [java] 2017-06-15 13:58:36,919 util.Log4JUtil                        INFO  - This log is stored in file: /Users/jake/Library/Preferences/JMRI/log/session.log [main]
     [java] 2017-06-15 13:58:36,927 apps.Apps                             INFO  - PanelPro version 4.9.1ish+jake+20170615T2058Z+R34e783c823 starts under Java 1.8.0_25 on Mac OS X x86_64 v10.12.5 at Thu Jun 15 13:58:36 PDT 2017 [main]
     [java] 2017-06-15 13:58:37,280 apps.Apps                             INFO  - Profiles not configurable. Using fallback per-application configuration. Error: org.xml.sax.SAXParseException; Premature end of file. [main]
     [java] 2017-06-15 13:58:37,384 node.NodeIdentity                     INFO  - Using jmri-784F4398C983-3f6e564e as the JMRI Node identity [main]
     [java] 2017-06-15 13:58:37,385 managers.JmriUserPreferencesManager   INFO  - No saved user preferences file [main]
     [java] 2017-06-15 13:58:37,397 apps.Apps                             INFO  - No saved preferences, will open preferences window.  Searched for /Users/jake/Library/Preferences/JMRI/PanelProConfig2.xml [main]
     [java] 2017-06-15 13:58:39,455 util.FileUtilSupport                  INFO  - File path program: is /Users/jake/Documents/Trains/JMRI/projects/JMRI/ [main]
     [java] 2017-06-15 13:58:39,456 util.FileUtilSupport                  INFO  - File path preference: is /Users/jake/Library/Preferences/JMRI/ [main]
     [java] 2017-06-15 13:58:39,456 util.FileUtilSupport                  INFO  - File path profile: is /Users/jake/Library/Preferences/JMRI/ [main]
     [java] 2017-06-15 13:58:39,457 util.FileUtilSupport                  INFO  - File path settings: is /Users/jake/Library/Preferences/JMRI/ [main]
     [java] 2017-06-15 13:58:39,458 util.FileUtilSupport                  INFO  - File path home: is /Users/jake/ [main]
     [java] 2017-06-15 13:58:39,459 util.FileUtilSupport                  INFO  - File path scripts: is /Users/jake/Documents/Trains/JMRI/projects/JMRI/jython/ [main]
     [java] 2017-06-15 13:58:41,353 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [init prefs]
     [java] java.util.ServiceConfigurationError: jmri.swing.PreferencesPanel: Provider jmri.web.server.RailroadNamePreferencesPanel could not be instantiated
     [java] 	at java.util.ServiceLoader.fail(ServiceLoader.java:232)
     [java] 	at java.util.ServiceLoader.access$100(ServiceLoader.java:185)
     [java] 	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:384)
     [java] 	at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
     [java] 	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
     [java] 	at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:180)
     [java] 	at apps.Apps$3.lambda$run$4(Apps.java:377)
     [java] 	at apps.Apps$3$$Lambda$47/2003770733.accept(Unknown Source)
     [java] 	at java.util.Optional.ifPresent(Optional.java:159)
     [java] 	at apps.Apps$3.run(Apps.java:376)
     [java] 	at java.lang.Thread.run(Thread.java:745)
     [java] Caused by: java.lang.NullPointerException: Profile must be specified.
     [java] 	at java.util.Objects.requireNonNull(Objects.java:228)
     [java] 	at jmri.beans.PreferencesBean.<init>(PreferencesBean.java:36)
     [java] 	at jmri.web.server.WebServerPreferences.<init>(WebServerPreferences.java:148)
     [java] 	at jmri.web.server.WebServerPreferences.lambda$getDefault$963(WebServerPreferences.java:155)
     [java] 	at jmri.web.server.WebServerPreferences$$Lambda$121/1394539280.get(Unknown Source)
     [java] 	at java.util.Optional.orElseGet(Optional.java:267)
     [java] 	at jmri.web.server.WebServerPreferences.getDefault(WebServerPreferences.java:154)
     [java] 	at jmri.web.server.RailroadNamePreferencesPanel.<init>(RailroadNamePreferencesPanel.java:23)
     [java] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
     [java] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
     [java] 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
     [java] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:408)
     [java] 	at java.lang.Class.newInstance(Class.java:438)
     [java] 	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:380)
     [java] 	... 8 more
     [java] 2017-06-15 13:58:46,922 managers.DefaultShutDownManager       INFO  - Normal termination complete [Thread-2]
     [java] Java Result: 126
```

Note that the PanelProConfig2.properties file is zero length:
```
-rw-r--r--   1 jake  staff     326 Jun 15 13:58 nodeIdentity.xml
drwxr-xr-x  17 jake  staff     578 Jun 15 13:58 preferences
-rw-r--r--   1 jake  staff       0 Jun 15 13:57 PanelProConfig2.properties
```

bobjacobsen, 2017-06-15:
Deleting the `PanelProConfig2.properties` file allowed an OK restart.

rhwood, 2017-06-15:
See #3452. I think the first crash occurred *while* writing PanelProConfig2.properties (which is where the starting profile is recorded for the next start).

commit b9224cb0 (2017-06-17): Merge pull request #3642 from rhwood/master

Fix issues with Profile selection
Fixes #3620 
Fixes #3641 
Fixes #3639 
Fixes #3452
closed 2017-06-17 by rhwood