JMRI/JMRI #3648: Create test release 4.7.7

bobjacobsen, 2017-06-17:
JMRI test release 4.7.7 is planned to be made from the release-4.76 branch plus plus critical fixes to new problems. It is tentatively scheduled for build on June 25, 2017, about a week from now. This is intended to be the last, clean-up test release before production release 4.8 in late June or early July. If it's not needed it will be skipped.

General PRs from this point forward will appear in JMRI 4.9.1, shortly after JMRI 4.8 is released at the end of June or early July. If you think a particular PR should be included in JMRI 4.7.7 instead, please add a note here.

If you're going to create something that you'd like to have in JMRI 4.7.7 and eventually release 4.8, please create your branch from the release-4.7.5 branch:
```
git fetch origin
git checkout origin/release-4.7.5
git branch -b my-new-branch-name
```

That can be easily merged via PR onto both `master` (for the main line of development) and release 4.7.7 and any successors.

rhwood, 2017-06-17:
Should I checkout release-4.7.5 or release-4.7.6 for a fix to get into 4.7.7?

bobjacobsen, 2017-06-17:
Either will work, but 4.7.5 is cleaner as it doesn't have a conflicting release.properties change. 

Bob

dheap, 2017-06-19:
#3657 needs to go in 4.7.7 and 4.8

rhwood, 2017-06-23:
Can #3671 (fixes #3626) and #3672 (correctly find Java 9 JRE) be considered for this release?

bobjacobsen, 2017-06-25:
See discussion in JMRI-developers list.  Decision is that there's enough outstanding that we should go back to 'master' for this, and have 4.7.8/4.7.9 be a restart of the stabilization process.

bobjacobsen, 2017-07-05:
4.7.7 has been built.

closed 2017-07-05 by bobjacobsen