JMRI/JMRI #3656: Single CV programmer not opening in 4.7.6

PUlvestad, 2017-06-18:
From users group:
Single CV Programmer won't open. I stepped down to 4.7.5, same issue. I then stepped down to 4.7.4,
it works. So something is broken for this between 4.7.4 and 4.7.5.

Alain LM

Here is a console print from the Single CV Programmer failure:

2017-06-18 13:27:25,701 util.Log4JUtil INFO - ****** JMRI log ******* [main]
2017-06-18 13:27:26,504 util.Log4JUtil INFO - This log is appended to file: /home/peter/.jmri/log/messages.log [main]
2017-06-18 13:27:26,506 util.Log4JUtil INFO - This log is stored in file: /home/peter/.jmri/log/session.log [main]
2017-06-18 13:27:26,612 apps.Apps INFO - PanelPro version 4.7.6+R570b2bc starts under Java 1.8.0_131 on Linux i386 v4.0.5-antix.1-486-smp at Sun Jun 18 13:27:26 MDT 2017 [main]
2017-06-18 13:27:32,885 apps.Apps INFO - Starting with profile JMRI.3eb5b6e2 [main]
2017-06-18 13:27:34,083 node.NodeIdentity INFO - Using jmri-00023F082735-3eb5b6e2 as the JMRI Node identity [main]
2017-06-18 13:27:38,534 loconet.LnPacketizer WARN - passing to xmit: unexpected exception: java.lang.NullPointerException [Thread-4]
2017-06-18 13:27:38,617 loconet.LnPacketizer WARN - passing to xmit: unexpected exception: java.lang.NullPointerException [Thread-5]
2017-06-18 13:27:38,730 hexfile.LnHexFilePort INFO - LocoNet Simulator Started [Thread-6]
2017-06-18 13:27:49,681 util.FileUtilSupport INFO - File path program: is /home/peter/JMRI/ [main]
2017-06-18 13:27:49,683 util.FileUtilSupport INFO - File path preference: is /home/peter/Dropbox/JMRI/ [main]
2017-06-18 13:27:49,697 util.FileUtilSupport INFO - File path profile: is /home/peter/.jmri/JMRI/ [main]
2017-06-18 13:27:49,699 util.FileUtilSupport INFO - File path settings: is /home/peter/.jmri/ [main]
2017-06-18 13:27:49,706 util.FileUtilSupport INFO - File path home: is /home/peter/ [main]
2017-06-18 13:27:49,714 util.FileUtilSupport INFO - File path scripts: is /home/peter/Dropbox/JMRI/Scripts/ [main]
2017-06-18 13:28:06,227 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.UnsupportedOperationException
at java.util.AbstractList.add(AbstractList.java:148)
at java.util.AbstractList.add(AbstractList.java:108)
at jmri.jmrit.progsupport.ProgOpsModePane.programmerSelected(ProgOpsModePane.java:240)
at jmri.jmrit.progsupport.ProgOpsModePane.<init>(ProgOpsModePane.java:209)
at jmri.jmrit.progsupport.ProgModePane.<init>(ProgModePane.java:69)
at jmri.jmrit.simpleprog.SimpleProgFrame.<init>(SimpleProgFrame.java:34)
at jmri.jmrit.simpleprog.SimpleProgAction.actionPerformed(SimpleProgAction.java:49)
at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
at java.awt.Component.processMouseEvent(Component.java:6533)
at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
at java.awt.Component.processEvent(Component.java:6298)
at java.awt.Container.processEvent(Container.java:2236)
at java.awt.Component.dispatchEventImpl(Component.java:4889)
at java.awt.Container.dispatchEventImpl(Container.java:2294)
at java.awt.Component.dispatchEvent(Component.java:4711)
at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
at java.awt.Container.dispatchEventImpl(Container.java:2280)
at java.awt.Window.dispatchEventImpl(Window.java:2746)
at java.awt.Component.dispatchEvent(Component.java:4711)
at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
at java.awt.EventQueue.access$500(EventQueue.java:97)
at java.awt.EventQueue$3.run(EventQueue.java:709)
at java.awt.EventQueue$3.run(EventQueue.java:703)
at java.security.AccessController.doPrivileged(Native Method)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
at java.awt.EventQueue$4.run(EventQueue.java:731)
at java.awt.EventQueue$4.run(EventQueue.java:729)
at java.security.AccessController.doPrivileged(Native Method)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Peter

AlanUS, 2017-06-18:
@PUlvestad 
Thanks for posting this. I was wondering if it was an issue on my local install, but apparently not.

dheap, 2017-06-18:
@AlanUS @PUlvestad I'll take a look at this as soon as I can. I added code here. It seems to be LocoNet-specific.

I had tested it with NCE Simulator, NCE Power Pro, NCE Power Cab and SPROG, but not with LocoNet Simulator.

 It seems to be LocoNet-specific. Are there any other connection types affected?

dheap, 2017-06-18:
#3657 fixes this. Should go in 4.7.7 and 4.8

AlanUS, 2017-06-19:
@dheap I'll make a test later today.
I observed this both with LocoNet simulator and Sprog IIv3.

dheap, 2017-06-19:
@AlanUS thanks. Let me know how it goes.

> I observed this both with LocoNet simulator and Sprog IIv3.

I had no problems with SPROG IIv3 when I was developing and testing it...

dheap, 2017-06-19:
@AlanUS You shouldn't have any more of this particular problem. Mea culpa, I was creating a reference to a returned array rather than a copy of the array and then adding elements. The LocoNet Simulator was returning an unmodifiable array.

AlanUS, 2017-06-20:
@dheap Was able to test just now, from current JMRI Master head. Works OK, both with LocoNet Simulator and Sprog IIv2. But according to what you fix, this was supposed to work anyway. Thanks.

closed 2017-06-20 by dheap