JMRI/JMRI #3660: FileNotFound exception when trying to open programmer

rm5248, 2017-06-20:
Situation: Starting up DecoderPro, trying to program a decoder.

I select the decoder in the list, and press the "Program" button.  An FileNotFound exception is then thrown, because it is looking for xml/programmers/Basic.xml.xml

Workaround: symlink Basic.xml.xml to Basic.xml, this allows JMRI to start and program, or go to Preferences->Roster->Programmer and change the format type.

I have seen this issue on multiple versions of JMRI, both 4.6 and 4.7.

Log attached.
[messages.log.txt](https://github.com/JMRI/JMRI/files/1089864/messages.log.txt)

dheap, 2017-06-21:
I can't reproduce this. I suspect it may be related to an existing badly stored preference file.

Does the problem come back if you change the format type beck from Advanced to Basic? I can't reproduce it either that way or with a new clean profile.

rm5248, 2017-06-21:
It looks like there was a bad file, changing it back seems to fix it.  However, one thing that I forgot to mention, is that when this happens the stack trace is printed out to the log, but there is no dialog presented to the user.  The 'Program' button simply doesn't work and there's no indication to the user that something has gone wrong.  You can reproduce that by re-naming the XML file and trying to re-load the programmer.

rhwood, 2017-06-22:
Renamed to reflect symptoms of issue as described:

> [W]hen this happens the stack trace is printed out to the log, but there is no dialog presented to the user. The 'Program' button simply doesn't work and there's no indication to the user that something has gone wrong.

closed 2021-04-14 by stale[bot]