JMRI/JMRI #3664: JMRI Wen

HugoFJ, 2017-06-21:


JamesWParker, 2018-07-21:
I have the same problem.  Has there been any progress and is there any workaround?

mstevetodd, 2018-07-21:
You're seeing this with 4.12?
It works fine for me. You can try it on my demo layout, http://jmri.mstevetodd.com:1080/panel/Layout/Waccamaw%20Coast%20Line
I think all signamlasts are set to Toggle Held, but I know the ones labeled East Yard work properly. I use that toggling to tell Dispatcher which AutoTrains are ready to run.
If you're at 4.12 and still see this issue, please attach your panel here and advise which Signal Mast.

JamesWParker, 2018-07-23:
I am running version 4.12+Rb6a9bb1 on a Raspberry Pi and both of the signal heads on the attached panel suffer the same problem (browser - Chrome or Edge both on Windows).  Any ideas are appreciated.

[Helston.zip](https://github.com/JMRI/JMRI/files/2221203/Helston.zip)

mstevetodd, 2018-07-24:
@HugoFJ are you still seeing the original problem reported here at v4.12? If you are, please attach a panel showing the issue, with recreation details. Also include what connections you are using. 
If not, I'm inclined to close this.

closed 2018-07-24 by mstevetodd

HugoFJ, 2018-07-24:
Hi Steve,

I need to confirm that I'm still experiencing this problem.

Will check and get back to you ASAP.

Regards,

Hugo

mstevetodd, 2018-07-25:
@JamesWParker I believe I have fixed your issue, and the new code is available in the latest Jenkins release (at http://jmri.tagadab.com/jenkins/job/Development/job/Packages/) and will be in future test and production releases. I would appreciate it if you'd give it a try and verify that it is fixed.

JamesWParker, 2018-07-26:
Thanks for this.  There seem to be quite a few releases here can you advise which one?

Many thanks

mstevetodd, 2018-07-26:
@JamesWParker the page http://jmri.tagadab.com/jenkins/job/Development/job/Packages/ has a section titled Last Successful Artifacts. There are three links, a Windows exe, a Mac dmg, and a Linux tgz. Choose the one that matches your operating system.

JamesWParker, 2018-07-28:
Thanks

I downloaded JMRI.4.13.1ish+jenkins+20180728T0556Z+Re026f29.  Unfortunately this seems to make the problem worse as the web view no longer reflects changes in the state made in the JMRI Panel

PR #5570 (2018-07-24): web panel, also skip dark state for clickmode 3 (#3664)
fixes #3664