JMRI/JMRI #3668: JMRI will not run with Java 9 on Windows

rhwood, 2017-06-23:
The JMRI launcher apparently looks for a registry entry that does not exist when Java 9 is installed to determine if the correct version of Java is installed.

See the thread at https://sourceforge.net/p/jmri/mailman/jmri-developers/thread/ae8940a8-5d85-a716-e1a3-87a411d0ef08%40upsys.co.uk/#msg35620522 for details.

A user has confirmed in private correspondence that JMRI 4.7.5 with Java 9+175 still has this problem.

rhwood, 2017-06-23:
> Here is the output from InstallTest.
 
CommandLine: LaunchJMRI  /debug /noisy apps.InstallTest.InstallTest        
Default options:
AppName: InstallTest
Class: apps.InstallTest.InstallTest
Parameters:
Noisy: 1
Force32bit: 0
Profile:
Testing for x64...
Result: 1
Setting x64 registry view...
Setting x86 registry view...
To copy this text to the clipboard, right click then choose
  'Copy Details To Clipboard'
Completed
 
What ithe output doesn’t show is that the test hangs at “setting x86 registry view”, displaying the “Java not found” message. When I dismiss the error box the test finishes.
 
Prior to the hang, the command prompt window had shown:
 
Testing for 32/64-bit Windows
 
Running on 64-bit Windows
 
Testing for 32/64-bit process
 
Running as a 64-bit process
 
Testing for Java working
 
java version "9"
Java(TM) SE Runtime Environment (build 9+175)
Java HotSpot(TM) 64-Bit Server VM (build 9+175, mixed mode)
 
Java is correctly working.
Press any key to continue . . .
 
Now testing JMRI
 
 
Brian

bobjacobsen, 2017-08-21:
The launcher in JMRI 4.9.3 has an update for this.

rhwood, 2017-09-03:
Works in current test release, so closing.

closed 2017-09-03 by rhwood