JMRI/JMRI #3678: OpenJDK and 4.7.6 freeze on opening

KenC57, 2017-06-25:
This is reported from a user after I gave the hint about using the Oracle versions instead. I suspect it is true for any using the new com library instead of RXTX. Here is his explanation of what he tried:

_
Well, after several painful days which have involved re-imaging my development machine to recover a working Debian 8 , then partitioning my machine's drive to support up to 4 different Linux systems, then loading the working Debian8, Ubuntu 17.04, and Debian9.0, I can finally compare the strange behavior that shows up on the newer installs.  On my Debian 8 machine, both here and at work, I installed Oracle Java 8 JRE.  Both of these machines run JMRI 4.7.6 without any big issues.  Both of the new installs use OpenJDK 8 JRE, and both have startup freezes with JMRI 4.7.6.

I just went through the setting up of Oracle Java 8 JRE on my Debian9 system, and it runs just fine, proving your suspicion that it might be a difference in the java implementations.  My own thought is that it would be well worth someone's effort to find a workable solution, because many users are NOT going to be happy with having to fiddle around with installing a "correct" java implementation when the expectation is that the OS packagers normally supply stuff that should just work.  What are the implications with Java 9?
Doug G
-

So I think one of the points, not sure what we can do, is many of the Linux packagers will load OpenJDK and not Oracle JDK by default. Either we need to figure out what is happening or at least figure the best way to document that users would need to add the Oracle JRE or JDK to make sure JMRI works.

rhwood, 2017-06-25:
If this user had installed 4.7.6 over the top of earlier versions of JMRI, JMRI failure is expected. We do not recommend installing newer versions of JMRI over older versions due to issues like this (although it may need more emphasis, the [JMRI Linux Install Guide](http://jmri.org/install/Linux.shtml) reads:

> Note: Each JMRI release is an independent set of files: It's intended to be used as a whole. Each release provides new files, and sometimes removes old files. You should therefore not expand the download into the same place as an existing JMRI installation. Instead, expand it into a separate location, and move in to it's final destination, completely replacing any previous version of JMRI.

However, that does not appear to be the case.

The Oracle vs OpenJDK Java implementation should be a non-issue since OpenJDK is the [reference implementation of Java](https://web.archive.org/web/20150906161310/https://blogs.oracle.com/henrik/entry/moving_to_openjdk_as_the) (see also [the 6th question on this FAQ](https://web.archive.org/web/20150910192908/https://blogs.oracle.com/henrik/entry/java_7_questions_answers) for a list of differences in Java 7). The most likely possibility is that Debian or Ubuntu deliberately removed a portion of the OpenJDK (or broke it into optional packages and not all were installed).

However, without logs from the failing JMRI runs, we cannot and should not make any decision about requiring the Oracle JDK.

rhwood, 2017-06-25:
I am unable to recreate a crash in JMRI 4.7.6 on CentOS 7 with OpenJDK 1.8.0_131-b12 as the only installed Java using a LocoBuffer-USB. My tests were limited to enabling/disabling track power and opening and closing some windows.

Please request logs of the failure from the user having issues. It appears to be unique to a specific JVM/OS/DCC Hardware combination.

rhwood, 2017-06-25:
(Initial comment updated to use ```>```s to delineate the original message instead of ```_```s.)

DaveF475, 2017-07-02:
I have a similar problem.

I had jmri-4.6 installed on my xubuntu-16.10 32 bit system. It worked fine.
Around June 1 I upgraded xubuntu to 17.04. Next time I tried to use jmri it
"froze" after lauching the first two or three windows. The exact behaviour
depended on whether I started DecoderPro or PanelPro. SIGKILL was the only
way to terminate it.

I thought that there might be a problem with my configuration file(s) so I
renamed my .jmri directory and tried a fresh start.
DecoderPro started as I expected and went through the setup wizard (I picked
Digitrax, loconet simulator). It droped me into the DecoderPro main window
from which I could add a new loco, save the roster, access help, close the
application. But if I try to do Edit->Preferences it will freeze.

If I close the first execution of DecoderPro and then launch it again with
the .jmri file it just created, it will freeze after displaying the main
window.

PanelPro always freezes after it displays its menu window.

The version of java it displays is
java 1.8.0_131(en_CA)
The package manager says that
openjdk-8-jre-8u131-b11-0ubuntu1.17.04.1 is installed and that it is the
latest version.

Today I renamed my JMRI directory to JMRI-4.6 and downloaded and installed
the 4.7.6 package. I deleted my .jmri directory and played with the new
jmri.  It behaves exactly the same as 4.6.  And, yes, it does identify
itself as 4.7.6.
When jmri is frozen, cpu utilization is 0, so it is wedged, not looping.

Following are the log files for a session that went:
rm .jmri
JMRI/DecoderPro
 add a loco, save, exit.
JMRI/DecoderPro
it froze, Killed it.
JMRI/PanelPro
It froze. Killed it. 

Messages.log:
2017-07-01 16:07:44,861 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-07-01 16:07:45,297 util.Log4JUtil                        INFO  - This log is appended to file: /home/fellows/.jmri/log/messages.log [main]
2017-07-01 16:07:45,304 util.Log4JUtil                        INFO  - This log is stored in file: /home/fellows/.jmri/log/session.log [main]
2017-07-01 16:07:45,361 apps.AppsBase                         INFO  - DecoderPro version 4.7.6+R570b2bc starts under Java 1.8.0_131 on Linux i386 v4.10.0-26-generic at Sat Jul 01 16:07:45 ADT 2017 [main]
2017-07-01 16:07:47,776 profile.ProfileManager                INFO  - Created default profile "My JMRI Railroad" [main]
2017-07-01 16:07:48,368 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3f755810 [main]
2017-07-01 16:07:48,933 node.NodeIdentity                     INFO  - Using jmri-002655C4EAD8-3f755810 as the JMRI Node identity [main]
2017-07-01 16:07:48,938 managers.JmriUserPreferencesManager   INFO  - No saved user preferences file [main]
2017-07-01 16:07:49,511 apps.AppsBase                         INFO  - No pre-existing config file found, searched for '/home/fellows/.jmri/My_JMRI_Railroad/ProfileConfig.xml' [main]
2017-07-01 16:07:49,514 util.FileUtilSupport                  INFO  - File path program: is /home/fellows/JMRI/ [main]
2017-07-01 16:07:49,515 util.FileUtilSupport                  INFO  - File path preference: is /home/fellows/.jmri/My_JMRI_Railroad/ [main]
2017-07-01 16:07:49,517 util.FileUtilSupport                  INFO  - File path profile: is /home/fellows/.jmri/My_JMRI_Railroad/ [main]
2017-07-01 16:07:49,520 util.FileUtilSupport                  INFO  - File path settings: is /home/fellows/.jmri/ [main]
2017-07-01 16:07:49,522 util.FileUtilSupport                  INFO  - File path home: is /home/fellows/ [main]
2017-07-01 16:07:49,524 util.FileUtilSupport                  INFO  - File path scripts: is /home/fellows/JMRI/jython/ [main]
2017-07-01 16:08:23,736 hexfile.LnHexFilePort                 ERROR - configure should not have been invoked [Start-Up Wizard Connect]
2017-07-01 16:08:23,739 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-8]
2017-07-01 16:08:24,053 logix.WarrantPreferences              ERROR - _fileName null. Could not create warrant preferences file. [Start-Up Wizard Connect]
2017-07-01 16:09:32,691 jmrit.XmlFile                         INFO  - No /home/fellows/.jmri/My_JMRI_Railroad/roster/foo.xml file to backup [AWT-EventQueue-1]
2017-07-01 16:09:32,759 jmrit.XmlFile                         INFO  - No /home/fellows/.jmri/My_JMRI_Railroad/roster.xml file to backup [AWT-EventQueue-1]
2017-07-01 16:10:13,022 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-1]
2017-07-01 16:10:22,936 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-07-01 16:10:23,176 util.Log4JUtil                        INFO  - This log is appended to file: /home/fellows/.jmri/log/messages.log [main]
2017-07-01 16:10:23,182 util.Log4JUtil                        INFO  - This log is stored in file: /home/fellows/.jmri/log/session.log [main]
2017-07-01 16:10:23,241 apps.AppsBase                         INFO  - DecoderPro version 4.7.6+R570b2bc starts under Java 1.8.0_131 on Linux i386 v4.10.0-26-generic at Sat Jul 01 16:10:23 ADT 2017 [main]
2017-07-01 16:10:25,068 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3f755810 [main]
2017-07-01 16:10:25,674 node.NodeIdentity                     INFO  - Using jmri-002655C4EAD8-3f755810 as the JMRI Node identity [main]
[... 45 lines cut ...]
[... cut ...]
2017-07-01 16:14:26,816 util.FileUtilSupport                  INFO  - File path profile: is /home/fellows/.jmri/My_JMRI_Railroad/ [main]
2017-07-01 16:14:26,828 util.FileUtilSupport                  INFO  - File path settings: is /home/fellows/.jmri/ [main]
2017-07-01 16:14:26,831 util.FileUtilSupport                  INFO  - File path home: is /home/fellows/ [main]
2017-07-01 16:14:26,850 util.FileUtilSupport                  INFO  - File path scripts: is /home/fellows/JMRI/jython/ [main]
----------------------

bobjacobsen, 2017-07-03:
It would help if somebody could get and append a thread dump in the "frozen" condition. That will (or at least might) give us a clue as to what's causing the hang.

DaveF475, 2017-07-05:
Following is the output of jstack against jmri-4.7.6 executing DecoderPro as describe in the comments above.  I can also provide the output of running jstack -m if that would be helpful.

Seems to be definitely a deadlock condition.

(I would like to attach these files, but when I try I just get an error message "Something went really wrong, and we can’t process that file. ")

==========jstack output===========
2017-07-04 09:25:59
Full thread dump OpenJDK Server VM (25.131-b11 mixed mode):

"Attach Listener" #38 daemon prio=9 os_prio=0 tid=0x77b73400 nid=0x21c2 waiting on condition [0x00000000]
   java.lang.Thread.State: RUNNABLE

"DestroyJavaVM" #37 prio=5 os_prio=0 tid=0xb6208000 nid=0x218a waiting on condition [0x00000000]
   java.lang.Thread.State: RUNNABLE

"Thread-6" #26 prio=5 os_prio=0 tid=0x737e3000 nid=0x21a4 waiting for monitor entry [0x738ad000]
   java.lang.Thread.State: BLOCKED (on object monitor)
	at java.awt.Component.addComponentListener(Component.java:5227)
	- waiting to lock <0x89a07020> (a jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane)
	at org.GNOME.Accessibility.AtkWrapper$3.eventDispatched(AtkWrapper.java:303)
	at java.awt.Toolkit$SelectiveAWTEventListener.eventDispatched(Toolkit.java:2425)
	at java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2316)
	at java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2316)
	at java.awt.Toolkit.notifyAWTEventListeners(Toolkit.java:2275)
	at java.awt.Component.dispatchEventImpl(Component.java:4785)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.Container.addImpl(Container.java:1138)
	- locked <0x8814e308> (a java.awt.Component$AWTTreeLock)
	at java.awt.Container.add(Container.java:973)
	at apps.gui3.TabbedPreferences$PreferencesCatItems$TabDetails.<init>(TabbedPreferences.java:522)
	at apps.gui3.TabbedPreferences$PreferencesCatItems.addPreferenceItem(TabbedPreferences.java:420)
	at apps.gui3.TabbedPreferences.addItem(TabbedPreferences.java:313)
	at apps.gui3.TabbedPreferences.addPreferencesPanel(TabbedPreferences.java:283)
	at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:190)
	- locked <0x8842b920> (a apps.gui3.TabbedPreferences)
	at apps.AppsBase$1.lambda$0(AppsBase.java:128)
	at apps.AppsBase$1$$Lambda$83/6576636.accept(Unknown Source)
	at java.util.Optional.ifPresent(Optional.java:159)
	at apps.AppsBase$1.run(AppsBase.java:127)
	at java.lang.Thread.run(Thread.java:748)

"Thread-5" #25 prio=5 os_prio=0 tid=0x737cc000 nid=0x21a3 waiting on condition [0x7345c000]
   java.lang.Thread.State: TIMED_WAITING (sleeping)
	at java.lang.Thread.sleep(Native Method)
	at jmri.jmrix.loconet.hexfile.LnHexFilePort.run(LnHexFilePort.java:90)
	at java.lang.Thread.run(Thread.java:748)

"LocoNet receive handler" #24 daemon prio=10 os_prio=0 tid=0x737cac00 nid=0x21a2 in Object.wait() [0x734ad000]
   java.lang.Thread.State: TIMED_WAITING (on object monitor)
	at java.lang.Object.wait(Native Method)
	- waiting on <0x8838a628> (a java.io.PipedInputStream)
	at java.io.PipedInputStream.read(PipedInputStream.java:326)
	- locked <0x8838a628> (a java.io.PipedInputStream)
	at java.io.PipedInputStream.read(PipedInputStream.java:377)
	- locked <0x8838a628> (a java.io.PipedInputStream)
	at java.io.DataInputStream.read(DataInputStream.java:149)
	at jmri.jmrix.loconet.LnPacketizer.readByteProtected(LnPacketizer.java:179)
	at jmri.jmrix.loconet.LnPacketizer$RcvHandler.run(LnPacketizer.java:342)
	at java.lang.Thread.run(Thread.java:748)

"LocoNet transmit handler" #23 daemon prio=9 os_prio=0 tid=0x737be400 nid=0x21a1 in Object.wait() [0x734fe000]
   java.lang.Thread.State: WAITING (on object monitor)
	at java.lang.Object.wait(Native Method)
	- waiting on <0x8838a838> (a jmri.jmrix.loconet.LnPacketizer$XmtHandler)
	at java.lang.Object.wait(Object.java:502)
	at jmri.util.WaitHandler.<init>(WaitHandler.java:67)
	- locked <0x8838a838> (a jmri.jmrix.loconet.LnPacketizer$XmtHandler)
	at jmri.jmrix.loconet.LnPacketizer$XmtHandler.run(LnPacketizer.java:505)
[... 178 lines cut ...]
	at jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane$$Lambda$149/23736962.propertyChange(Unknown Source)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at jmri.beans.Bean.lambda$6(Bean.java:154)
	at jmri.beans.Bean$$Lambda$19/3728767.run(Unknown Source)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at org.GNOME.Accessibility.AtkWrapper$5.dispatchEvent(AtkWrapper.java:700)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Found 1 deadlock

=============End of jstack output

bobjacobsen, 2017-07-05:
Thanks.  Definitely a deadlock.  The problem is in "thread-6". That's running Swing code off the Swing thread.

Thread-6 is probably being launched around line 121-135 of apps.AppsBase to initialize the preferences GUI.  AppsBase line 128 calls init() is a TabbedPreferences object (specifically `apps.gui3.TabbedPreferences`) which is a _synchronized_ method.  So there's the first lock.  It then tries get back to AWT/Swing to add a Component (line 522 of TabbedPreferences), which sets a second, overlapping AWT lock.  That add() invokes listeners _not_ on the Swing thread, because the add() wasn't invoked on the Swing thread, which causes Swing to try to get a lock on jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane (a Swing subclass) object off the Swing thread.  But that's not available, so it hangs, and  the Swing thread hangs too.

It looks like the possibilities are:
- Make sure that the synchronization in UserMessagePreferencesPane and TabbedPreferences is right. I don't see anything wrong, but I haven't studied in detail, and overlapping locks are tricky
- Make sure that the Preferences GUI isn't displaying anything.  If the preferences have never been displayed, this should be safe, at least from the Swing side. Maybe it's timing related, as it depends on whether preferences initialization happens before something causes the preferences to display?
- But really the solution is to push all the Swing initialization onto the Swing thread.  If it's too monolithic and stops event processing, then it'll need to be broken up into reasonable-sized runOnGUI calls.

bobjacobsen, 2017-07-09:
Could somebody who can reproduce this please try the development packages in http://jmri.tagadab.com/jenkins/job/Development/job/Packages/1660/ ? There's a threading change there from #3734 that might fix it.

DaveF475, 2017-07-10:
Tried it. Sadly, behaviour is basically the same.  I am attaching the jstack outputs. And also the messages.log.  The messages log records 3 sessions: 

- The first invoked DecoderPro with no pre-existing .jmri directory.  It worked and created the directory. My impression is that it was more stable than the 4.7.6 equivalent session.  I tabbed about  a bit - things seemed to go as expected. I was able to close this session normally.
- The second was a PanelPro invocation.  It froze as soon as it displayed its main window. This is what the stack traces record.
- The thrd was a second DecoderPro invocation.  It froze after displaying its, empty, main window.  Same behaviour as before.

[jstack.txt](https://github.com/JMRI/JMRI/files/1134180/jstack.txt)
[jstack-m.txt](https://github.com/JMRI/JMRI/files/1134193/jstack-m.txt)
[messages.txt](https://github.com/JMRI/JMRI/files/1134195/messages.txt)

bobjacobsen, 2017-07-10:
Thank you, @DaveF475!  

Is that top [jstack.txt](https://github.com/JMRI/JMRI/files/1134180/jstack.txt) from the updated JMRI version?  The line numbers seem to be exactly the same as previous.  Perhaps you attached a file from the earlier test by accident?

DaveF475, 2017-07-10:
I don't think I picked up a wrong file.  I just downloaded the file from here and compared it with what i produced last night - no difference.

To be sure I repeated the PanelPro execution. Verified it is your test version id that is displayed. It is. It is frozen. The messages.log and jstack traces are attached. (I deleted the old  messages.log file before running the test.)

[jstack-panelpro-2.txt](https://github.com/JMRI/JMRI/files/1135700/jstack-panelpro-2.txt)
[messages-panelpro-2.txt](https://github.com/JMRI/JMRI/files/1135696/messages-panelpro-2.txt)

costing, 2017-07-14:
For the record, the problem seems to still be there in 4.7.8:

```
$ ./DecoderPro
...
2017-07-14 10:49:52,158 apps.AppsBase                         INFO  - DecoderPro version 4.7.8+R583bddd starts under Java 1.8.0_131 on Linux amd64 v4.10.0-26-generic at Fri Jul 14 10:49:52 CEST 2017 [main]
```

```
Found one Java-level deadlock:
=============================
"Thread-22":
  waiting to lock monitor 0x00007fa1c0034598 (object 0x00000005cca7f618, a jmri.jmrit.beantable.usermessagepreferences.UserMessagePreferencesPane),
  which is held by "AWT-EventQueue-1"
"AWT-EventQueue-1":
  waiting to lock monitor 0x00007fa20c004328 (object 0x00000005ca800c90, a java.awt.Component$AWTTreeLock),
  which is held by "Thread-22"
```

CoolGames, 2017-12-18:
Seems like I am in this effort to fix.
I am NOT a programmer but a retired applications engineer.
I am back with Ubuntu Studio 17.10 and unzipped JMRI.4.9.8.R732d507.tgz.

$ java -version
'openjdk version "1.8.0_151"
OpenJDK Runtime Environment (build 1.8.0_151-8u151-b12-0ubuntu0.17.10.2-b12)
OpenJDK 64-Bit Server VM (build 25.151-b12, mixed mode)

I will be using ESU ECoS for station and ESU Programmer for sound 
and Digitrax Zephyr for NMRA decoders.

I installed Eclipse 3.8.1 but can not get git and GitHub as import path (yet).

I am now watching this issue as well.

bobjacobsen, 2017-12-19:
@balazsracz found another case, log and thread trace attached.

[init-deadlock-4.9.8.txt](https://github.com/JMRI/JMRI/files/1572270/init-deadlock-4.9.8.txt)

balazsracz, 2017-12-21:
@bobjacobsen 
Steps to repro (my bug):
- Download the Debian live CD image from https://cdimage.debian.org/debian-cd/current-live/amd64/bt-hybrid/debian-live-9.3.0-amd64-gnome.iso.torrent Other versions and download options are available via https://www.debian.org/CD/live/
- Install Oracle VirtualBox on your computer
- Create a virtual machine with a CD drive that is linked to the above ISO. It is not necessary to have a hard disk set up, or you can leave it at the default.
- Start the virtual machine
- In the VM, download JMRI and install it (decompress). To get a terminal, click the "activities" button on the top left and start typing "terminal" into the search box; then click on the resulting icon.
- start the PanelPro application
- after the application is frozen (will be at the second splash screen), you can examine the java process, or get the stacktrace with killall -3 java
- Be aware that if the virtual machine is shut down, everything you did inside the VM will be lost.

bobjacobsen, 2017-12-22:
Thank you!  That link got me an image that shows the problem.

MonitorComputing, 2018-01-15:
If I might make a "me to" post I've just loaded the JMRI 4.10 release onto a laptop running Debian 9 (Stretch) with OpenJDK 8. This hangs on opening the main PanelPro screen, usually with the screen blank but once with it fully populated. I've attached a text file with machine, OS, and JDK info along with the thread dump reporting one deadlock.
[deadlock.log](https://github.com/JMRI/JMRI/files/1632797/deadlock.log)

MonitorComputing, 2018-01-15:
Guess what. Just rolled up to OpenJDK 9 and the problem seems to have gone away :-)

wilkenw, 2018-02-05:
I want to thank all who have posted on this subject.  It provided the badly needed clue to solve hours of frustration in an effort to get JMRI running.  In my case, I am running JMRI 4.10 on a Macbook Pro running Ubuntu 17.10 with a USB connection to an NCE PowerCab.  After installing Oracle Java (OpenJDK Runtime Environment (build 1.8.0_151-8u151-b12-0ubuntu0.17.10.2-b12), and configuring CP2102N, my session.log was filled with errors advising that ttyUSB0 was in use.  A simple "ls -l /dev/ttyUSB0" check indicated that the problem was permissions, which I "fixed" with a "chmod 666" on the fly.  At which point, I could create a "railroad" and add a loco to the roster.  But it was impossible to run the loco; moreover, any subsequent attempts to run DecoderPro or PanelPro resulted in frozen screens that could be killed only by logging off my system.  Not good.  But once I read about your Java woes with other versions of linux, I removed all the pieces I had download from Oracle, then used the Synaptic package manager to install openjdk-09.  Voila, magic:  no more screen freezes.

PR #3734 (2017-07-09): Threading fix
Move some preferences operations to the Swing thread in an attempt to resolve #3678
closed 2021-04-14 by stale[bot]