JMRI/JMRI #3688: UncaughtException creating new Roster Index with locomotive id="99999999999999"

dheap, 2017-06-28:
Had an UncaughtException creating a new Roster Index (no existing roster.xml) with locomotive id="99999999999999"
Tried several times and then JMRI refused to quit.

> 2017-06-28 11:53:35,529 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
> java.lang.NumberFormatException: For input string: "99999999999999"
> 	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
> 	at java.lang.Integer.parseInt(Integer.java:583)
> 	at java.lang.Integer.valueOf(Integer.java:766)
> 	at jmri.util.PreferNumericComparator.compare(PreferNumericComparator.java:27)
> 	at javax.swing.DefaultRowSorter.compare(DefaultRowSorter.java:968)
> 	at javax.swing.DefaultRowSorter.access$100(DefaultRowSorter.java:112)
> 	at javax.swing.DefaultRowSorter$Row.compareTo(DefaultRowSorter.java:1376)
> 	at javax.swing.DefaultRowSorter$Row.compareTo(DefaultRowSorter.java:1366)
> 	at java.util.ComparableTimSort.binarySort(ComparableTimSort.java:262)
> 	at java.util.ComparableTimSort.sort(ComparableTimSort.java:207)
> 	at java.util.Arrays.sort(Arrays.java:1246)
> 

Tried again later and couldn't reproduce. Full log and offending roster entry attached.

[99999999999999.zip](https://github.com/JMRI/JMRI/files/1107426/99999999999999.zip)

dheap, 2021-03-31:
Will check this one.

bobjacobsen, 2021-03-31:
99999999999999 is too big to be an integer value.

dheap, 2021-03-31:
> 99999999999999 is too big to be an integer value.

Agreed, but since ID is a text field any combination of alphanumerics should be acceptable. I realise the integer parsing is part of the sort algorithm but it should ignore the unparseable and treat it as text.

dheap, 2021-05-16:
See https://github.com/JMRI/JMRI/issues/3688#issuecomment-811435788

bobjacobsen, 2021-05-17:
@dheap made a good suggestion on how to handle this. I'll PR a fix for this shortly.

PR #9798 (2021-05-18): Fall back to text compare for too-big numbers
Fixes #3688
closed 2021-05-18 by danielb987