JMRI/JMRI #3706: DecoderPro: OutOfBounds exception with Uhlenbrock LocoNet Messages

michael71, 2017-07-02:
when trying to program a decoder (decoder type D&H DH22A9, JMRI connected to IB-Com), I get a lot of "ArrayIndexOutOfBounds" exceptions and cannot even read the data of the decoder - the "UhlenbrockPacketizer" seems to get messages which it cannot interpret because of wrong message lengths. (see output from messages.log below and the command line output using JMRI version is 4.7.5).

Are there any known limitations with Uhlenbrock equipment? There is nothing mentioned on
[http://jmri.org/help/en/html/hardware/loconet/Uhlenbrock.shtml](
http://jmri.org/help/en/html/hardware/loconet/Uhlenbrock.shtml)

thanks for your help, 
with regards
Michael.

------ messages.log
2017-06-30 15:13:28,185 uhlenbrock.UhlenbrockPacketizer WARN - LocoNet message with opCode: b0 ended early. Byte2 is also an opcode: b0 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:13:48,212 uhlenbrock.UhlenbrockPacketizer WARN - Ignore Loconet packet with bad checksum: B0 7D 01 27 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:13:48,213 uhlenbrock.UhlenbrockPacketizer WARN - run: unexpected LocoNetMessageException: jmri.jmrix.loconet.LocoNetMessageException [LocoNet Uhlenbrock receive handler]
2017-06-30 15:14:08,211 uhlenbrock.UhlenbrockPacketizer ERROR - LocoNet message length invalid: 1 opcode: ed [LocoNet Uhlenbrock receive handler]
2017-06-30 15:14:08,212 loconet.LocoNetMessage ERROR - reference element 1 in message of 1 elements: ED [LocoNet Uhlenbrock receive handler]
2017-06-30 15:14:08,212 uhlenbrock.UhlenbrockPacketizer WARN - run: unexpected Exception: java.lang.ArrayIndexOutOfBoundsException: 1 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:14:28,213 uhlenbrock.UhlenbrockPacketizer ERROR - LocoNet message length invalid: 1 opcode: e5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:14:28,214 loconet.LocoNetMessage ERROR - reference element 1 in message of 1 elements: E5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:14:28,215 uhlenbrock.UhlenbrockPacketizer WARN - run: unexpected Exception: java.lang.ArrayIndexOutOfBoundsException: 1 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:14:48,213 uhlenbrock.UhlenbrockPacketizer ERROR - LocoNet message length invalid: 1 opcode: ed [LocoNet Uhlenbrock receive handler]
2017-06-30 15:14:48,214 loconet.LocoNetMessage ERROR - reference element 1 in message of 1 elements: ED [LocoNet Uhlenbrock receive handler]
2017-06-30 15:14:48,214 uhlenbrock.UhlenbrockPacketizer WARN - run: unexpected Exception: java.lang.ArrayIndexOutOfBoundsException: 1 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:15:02,930 uhlenbrock.UhlenbrockPacketizer ERROR - LocoNet message length invalid: 1 opcode: e5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:15:02,930 loconet.LocoNetMessage ERROR - reference element 1 in message of 1 elements: E5 [LocoNet Uhlenbrock receive handler]

----- java ERRORS on the commandline
java.lang.ArrayIndexOutOfBoundsException: 1
at jmri.jmrix.loconet.LocoNetMessage.setElement(LocoNetMessage.java:114)
at jmri.jmrix.loconet.uhlenbrock.UhlenbrockPacketizer$RcvHandler.run(UhlenbrockPacketizer.java:179)
at java.lang.Thread.run(Thread.java:748)
2017-06-30 15:16:48,218 uhlenbrock.UhlenbrockPacketizer ERROR - LocoNet message length invalid: 1 opcode: e5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:16:48,218 loconet.LocoNetMessage ERROR - reference element 1 in message of 1 elements: E5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:16:48,219 uhlenbrock.UhlenbrockPacketizer WARN - run: unexpected Exception: java.lang.ArrayIndexOutOfBoundsException: 1 [LocoNet Uhlenbrock receive handler]
java.lang.ArrayIndexOutOfBoundsException: 1
at jmri.jmrix.loconet.LocoNetMessage.setElement(LocoNetMessage.java:114)
at jmri.jmrix.loconet.uhlenbrock.UhlenbrockPacketizer$RcvHandler.run(UhlenbrockPacketizer.java:179)
at java.lang.Thread.run(Thread.java:748)
2017-06-30 15:17:10,255 uhlenbrock.UhlenbrockPacketizer WARN - LocoNet message with opCode: b4 ended early. Expected length: 4 seen length: 2 unexpected byte: e7 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:17:28,228 uhlenbrock.UhlenbrockPacketizer WARN - LocoNet message with opCode: e7 ended early. Expected length: 124 seen length: 22 unexpected byte: e5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:17:38,251 uhlenbrock.UhlenbrockPacketizer ERROR - LocoNet message length invalid: 1 opcode: e5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:17:38,251 loconet.LocoNetMessage ERROR - reference element 1 in message of 1 elements: E5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:17:38,251 uhlenbrock.UhlenbrockPacketizer WARN - run: unexpected Exception: java.lang.ArrayIndexOutOfBoundsException: 1 [LocoNet Uhlenbrock receive handler]
java.lang.ArrayIndexOutOfBoundsException: 1
at jmri.jmrix.loconet.LocoNetMessage.setElement(LocoNetMessage.java:114)
at jmri.jmrix.loconet.uhlenbrock.UhlenbrockPacketizer$RcvHandler.run(UhlenbrockPacketizer.java:179)
at java.lang.Thread.run(Thread.java:748)
2017-06-30 15:17:58,246 uhlenbrock.UhlenbrockPacketizer WARN - LocoNet message with opCode: e7 ended early. Expected length: 124 seen length: 8 unexpected byte: e5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:18:08,269 uhlenbrock.UhlenbrockPacketizer ERROR - LocoNet message length invalid: 1 opcode: e5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:18:08,270 loconet.LocoNetMessage ERROR - reference element 1 in message of 1 elements: E5 [LocoNet Uhlenbrock receive handler]
2017-06-30 15:18:08,270 uhlenbrock.UhlenbrockPacketizer WARN - run: unexpected Exception: java.lang.ArrayIndexOutOfBoundsException: 1 [LocoNet Uhlenbrock receive handler]
java.lang.ArrayIndexOutOfBoundsException: 1
at jmri.jmrix.loconet.LocoNetMessage.setElement(LocoNetMessage.java:114)
at jmri.jmrix.loconet.uhlenbrock.UhlenbrockPacketizer$RcvHandler.run(UhlenbrockPacketizer.java:179)
at java.lang.Thread.run(Thread.java:748)
2017-06-30 15:18:28,270 uhlenbrock.UhlenbrockPacketizer WARN - LocoNet message with opCode: a0 ended early. Expected length: 4 seen length: 2 unexpected byte: ed [LocoNet Uhlenbrock receive handler]
2017-06-30 15:18:28,271 uhlenbrock.UhlenbrockPacketizer ERROR - LocoNet message length invalid: 1 opcode: ed [LocoNet Uhlenbrock receive handler]
2017-06-30 15:18:28,271 loconet.LocoNetMessage ERROR - reference element 1 in message of 1 elements: ED [LocoNet Uhlenbrock receive handler]
2017-06-30 15:18:28,271 uhlenbrock.UhlenbrockPacketizer WARN - run: unexpected Exception: java.lang.ArrayIndexOutOfBoundsException: 1 [LocoNet Uhlenbrock receive handler]
java.lang.ArrayIndexOutOfBoundsException: 1
at jmri.jmrix.loconet.LocoNetMessage.setElement(LocoNetMessage.java:114)
at jmri.jmrix.loconet.uhlenbrock.UhlenbrockPacketizer$RcvHandler.run(UhlenbrockPacketizer.java:179)
at java.lang.Thread.run(Thread.java:748)

rhwood, 2017-07-02:
Original report edited for formatting by adding code markers (three backticks) around output.

bobjacobsen, 2017-07-02:
The `LocoNet message with opCode: NN ended early` messages are the root problem.  JMRI is seeing incorrect messages coming from the LocoNet.  The most likely cause is characters being lost.

How do you have JMRI connected to the LocoNet?  What hardware, and how is that configured?

michael71, 2017-07-02:
I have an IB-COM connected - via USB, Baudrate setting 115200, LN-Direct-Mode=Off

michael71, 2017-07-02:
Thanks for your question, Bob,
I never thought about a non-working USB connection ... But I now connected the PC via a LoconetUSB-Interface (Uhlenbrock 63120) and NOT via the IB-Com directly - and the errors disappeared !  
Well, pretty strange. How can this happen?

bobjacobsen, 2017-07-03:
The Intellibox has a tiny buffer for serial communications, and (sometimes) it doesn't manage it well.  You should probably just stick with the LocoNetUSB. But if you want to try the IB again, try a lower baud rate.

devel-bobm, 2017-09-21:
@michael71, I wonder your "communications problems" relates to "hardware flow control"...  There have been some recent changes to "fix" JMRI's "flow control" features on "serial connections", and this may relate to your problem.  See "Pull Request" #4041.  This fix is applied in the recent JMRI 4.9.4 "test release".

michael71, 2017-09-23:
thanks,  Bob, I just downloaded 4.9.4 and tried to start it. 
However, I cannot connect to my IBCom (USB Interface) with that version, there is the following ERROR:

 jmrix.AbstractSerialPortController    ERROR - no match to (115200) in currentBaudNumber

? 115200 is the "baudrate" which is configure in the IB-COM (checked with Uhlenbrocks config program)

the complete log:
2017-09-23 19:00:03,191 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2017-09-23 19:00:03,219 util.Log4JUtil                        INFO  - This log is appended to file: /home/mblank/.jmri/log/messages.log [main]
2017-09-23 19:00:03,220 util.Log4JUtil                        INFO  - This log is stored in file: /home/mblank/.jmri/log/session.log [main]
2017-09-23 19:00:03,226 apps.AppsBase                         INFO  - DecoderPro version 4.9.4+R3bcfc65 starts under Java 1.8.0_144 on Linux amd64 v4.4.0-96-generic at Sat Sep 23 19:00:03 CEST 2017 [main]
2017-09-23 19:00:06,114 gui3.Apps3                            INFO  - Starting with profile Meine_JMRI_Bahngesellschaft.3f09930f [main]
2017-09-23 19:00:06,247 node.NodeIdentity                     INFO  - Using jmri-2C4138995ED9-3f09930f as the JMRI Node identity [main]
2017-09-23 19:00:06,453 jmrix.AbstractSerialPortController    ERROR - no match to (115200) in currentBaudNumber [main]
2017-09-23 19:00:06,457 jmrix.ConnectionConfigManager         ERROR - Unable to load [Element: <connection/>] into jmri.jmrix.loconet.uhlenbrock.configurexml.ConnectionConfigXml [main]
purejavacomm.PureJavaIllegalStateException: JTermios call returned -1 at class purejavacomm.PureJavaSerialPort line 532
	at purejavacomm.PureJavaSerialPort.checkReturnCode(PureJavaSerialPort.java:1305)
	at purejavacomm.PureJavaSerialPort.setSerialPortParams(PureJavaSerialPort.java:532)
	at jmri.jmrix.loconet.uhlenbrock.UhlenbrockAdapter.setSerialPort(UhlenbrockAdapter.java:89)
	at jmri.jmrix.loconet.locobuffer.LocoBufferAdapter.openPort(LocoBufferAdapter.java:100)
	at jmri.jmrix.configurexml.AbstractSerialConnectionConfigXml.load(AbstractSerialConnectionConfigXml.java:97)
	at jmri.jmrix.ConnectionConfigManager.initialize(ConnectionConfigManager.java:106)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:263)
	at jmri.implementation.JmriConfigurationManager.lambda$3(JmriConfigurationManager.java:259)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:258)
	at jmri.implementation.JmriConfigurationManager.lambda$3(JmriConfigurationManager.java:259)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:258)
	at jmri.implementation.JmriConfigurationManager.lambda$1(JmriConfigurationManager.java:183)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:182)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:170)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:160)
	at apps.AppsBase.setAndLoadPreferenceFile(AppsBase.java:281)
	at apps.gui3.Apps3.setAndLoadPreferenceFile(Apps3.java:367)
	at apps.AppsBase.<init>(AppsBase.java:106)
	at apps.gui3.Apps3.<init>(Apps3.java:92)
	at apps.gui3.dp3.DecoderPro3.<init>(DecoderPro3.java:27)
	at apps.gui3.dp3.DecoderPro3.main(DecoderPro3.java:101)
2017-09-23 19:00:06,461 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.jmrix.ConnectionConfigManager: Unable to create connection "LocoNet" (L). [main]
2017-09-23 19:00:06,500 loconet.LnPowerManager                ERROR - Power Manager Created, yet there is no Traffic Controller [main]
2017-09-23 19:00:06,501 jmri.InstanceManager                  ERROR - Should not set default of type jmri.CommandStation to null value [main]
2017-09-23 19:00:06,502 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for LocoNet (class jmri.jmrix.loconet.uhlenbrock.UhlenbrockSystemConnectionMemo) provides a null interface jmri.CommandStation instance [main]
2017-09-23 19:00:06,503 jmri.InstanceManager                  ERROR - Should not set default of type jmri.ThrottleManager to null value [main]
2017-09-23 19:00:06,503 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for LocoNet (class jmri.jmrix.loconet.uhlenbrock.UhlenbrockSystemConnectionMemo) provides a null interface jmri.ThrottleManager instance [main]
2017-09-23 19:00:06,504 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.managers.ManagerDefaultSelector: System connection LocoNet provides a null manager for interface jmri.ThrottleManager [main]
2017-09-23 19:00:06,655 plementation.JmriConfigurationManager ERROR - Exception initializing apps.StartupActionsManager: jmri.util.prefs.InitializationException: Unable to run startup actions due to earlier failures. [main]

devel-bobm, 2017-09-23:
@michael71 , I see one thing that is likely the indicator of the source of your problem:

`jmrix.AbstractSerialPortController ERROR - no match to (115200) in currentBaudNumber [main]`

Your log seems to indicate that an improper value for the "baud rate" has been passed to the serial communication library.  But I don't see how that can happen in the code.  I'm sure I'm missing something...

But I instead expect that using an "old-style" JMRI profile with this new code might not work correctly.

Can you post a copy of your configuration profile .xml file to this issue?

If this is indeed the source of the a problem, then creating a new configuration profile ought to solve the problem.  Can you try creating a new profile with 4.9.4 and checking to see if it will properly initialize the hardware, and report back to us?

devel-bobm, 2017-09-24:
@michael71,
Ignore my previous comments.

I find that creation of an initial profile generates "incorrect data" in the profile .xml file for the baud rate, due to incorrect and/or incomplete "internationalization" in the Uhlenbrock code.

The easy workaround for your problem is to open the profile (and get the error), then edit the preferences and change the connection's baud rate to the correct value.  Save the updated profile.  Then quit and re-start JMRI.  This should solve the .XML file error and allow communication with the Uhlenbrock hardware.

devel-bobm, 2017-09-24:
@bobjacobsen , the root-cause analysis is that Uhlenbrock connection profiles are saving Locale-specific coding for the baud rate, but the "default" baud rate uses a simple numeric string which is not Locale-specific.  If the user crates a new profile without selecting the baud rate, the "default" baud rate string is saved to the profile.  

From `loconet.Uhlenbrock.UhlenbrockAdapter.java:35`:

`        configureBaudRate("115200"); //Set the default baud rate`

Upon reading the profile, the baud rate string does not match any of the Locale-specific baud-rate values in `validSpeeds[]`, and therefore fails to look-up the corresponding integer value from `validSpeedValues[]`.

Is there a generic assumption that the JMRI configuration profile will contain non-Locale-specific text for things like baud rates?  If so, then this connection needs to be "fixed" with respect to I18N.  And something in JMRI's XML persistence and I18N web pages needs to define the "rules" for configuration profile and I18N.

silverailscolo, 2017-09-24:
In the GUI the as-stored values to me looked a bit hard to read (for the high numbers) so I tried using Formatting to improve that. In mid July I separated the values used in the code from the display (taken from Bundles), linked by the keys. I saw no way to i18n without that showing up in the profile.
The interaction with new feautures and storing/loading older panels is still an issue. Changing your Locale and reloading a panel provides similar issues, eg. see #1745
Now that we know, I'll write a line or 2 in the Release Notes. A 'rule' would be helpful, but until we have the code fully separated from the display, it could also limit the path forward.

closed 2021-04-14 by stale[bot]