JMRI/JMRI #3711: Linux menu integration

bobjacobsen, 2017-07-03:
From #1978:

> there should be an integration of JMRI in the program menu on systems like Ubuntu or OpenSuse to allow non technical persons to start it.

This might just be a help-page discussion of how to add JMRI application(s) to those program menus. Or might involved modified or custom-use startup scripts.

Dylan-M, 2017-10-27:
@bobjacobsen I can base it on how some other software (Like Eclipse) handles it and include a .desktop file (or files) that takes care of this if you'd like to assign it to me.

bobjacobsen, 2017-10-28:
@Dylan-M Excellent!  Thanks.  

I don't know enough about .desktop files to know how best to distribute those.  Do they need to be merged with existing files?

Dylan-M, 2017-10-29:
Likely I'll be doing a script that will create the .desktop files. I'm not sure what you mean by merged with existing files... each program gets it's own .desktop file, sometimes more than one (like JMRI will if I do it as I'm planning). If you mean do they have to be put in a specific place, then yes. There (as with many things on Linux) are a few places they can go. I'll make sure to well document this and make the script able to put things into place automatically.

closed 2021-04-14 by stale[bot]