JMRI/JMRI #3731: DCCppOverTcpServer is not started by JmriFaceless

costing, 2017-07-08:
Hi,

I'm trying to automatically start DCC++ server on an RPI that would normally run JmriFaceless. If I start it interactively (DecoderPro) the DCC++ server is automatically started, so the configuration seems ok. But JmriFaceless doesn't seem to reach the same point, as the following lines only show up when running DecoderPro but not for JmriFaceless:

```
2017-07-08 18:57:50,043 dccppovertcp.Server                   INFO  - Starting new DCCppOverTcpServer listener on port 1235 [main]
2017-07-08 18:57:50,076 dccppovertcp.Server                   INFO  - Starting ZeroConfService _dccppovertcpserver._tcp.local for DCCppOverTCP Server [main]
```

WiThrottle and the web server are started ok, just the DCC++ server is missing.

Thanks a lot for looking into this!

.costin

mstevetodd, 2017-07-08:
Hi @costing ,
Are you running JmriFaceless with the same profile as your interactive DecoderPro? If not, you'll need to use the same profile, or move the action over to the one used by JmriFaceless. The actions are stored as <perform....> elements of ProfileConfig.xml

BTW, how do you start the DCCppOverTcpServer? I don't see it under "Perform Action..." in Preferences. 

Regards,
  SteveT

costing, 2017-07-09:
Hi Steve,

Yes, same and only profile. I start the server according to the [documentation](http://jmri.org/help/en/html/hardware/dccpp/index.shtml) # Connecting to a DCC++ System using DCC++ Over TCP. So DecoderPro main menu bar -> "DCC++" -> "Load DCC++ over TCP server". From that window I select "Start Server at Application Startup" and "Save Settings". This has the effect of generating [profile]/DCCppOverTcpSettings.ini where it writes "AutoStart = 1". But no .xml is touched. And yes, I don't see it under "Perform Action..." either.

Cheers,

.costin

mstevetodd, 2017-07-10:
@costing ,
JmriFaceless isn't reading the .ini, just the normal <perform...> actions. Someone will need to convert DCCppOverTcp to show in Perform Action for this to work.
--SteveT

costing, 2017-07-10:
Hi Steve,

The pieces seem to be there and just this entry was missing from the Perform Actions list:
```
diff --git a/java/src/apps/ActionListBundle.properties b/java/src/apps/ActionListBundle.properties
index ca75bc9111..f3356900c4 100644
--- a/java/src/apps/ActionListBundle.properties
+++ b/java/src/apps/ActionListBundle.properties
@@ -82,6 +82,9 @@ jmri.jmrix.libusb.UsbViewAction = Open USB Device Viewer
 # web items
 jmri.web.server.WebServerAction = Start Web Server

+# DCC++ TCP Server
+jmri.jmrix.dccpp.dccppovertcp.ServerAction = Start DCC++ TCP Server
```

It's not pretty in the sense that it would like to display the frame at startup and this produces an exception when run under JmriFaceless but it works for my purpose of running it without an interface :) I let you decide if this is the correct way of addressing this issue or not.

Cheers,

.costin

costing, 2017-07-23:
Commit 52dbdf7 is ok for me, please close the issue if you think the following exception in JmriFaceless is not a problem (the server thread I need runs just fine).
```
2017-07-23 21:53:05,925 startup.AbstractActionModel           ERROR - Error while performing startup action for class: jmri.jmrix.dccpp.dccppovertcp.ServerAction [main]                                                                     
java.awt.HeadlessException                                                                                                                                                                                                                   
        at java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:204)                                                                                                                                                          
        at java.awt.Window.<init>(Window.java:536)                                                                                                                                                                                           
        at java.awt.Frame.<init>(Frame.java:420)                                                                                                                                                                                             
        at java.awt.Frame.<init>(Frame.java:385)                                                                                                                                                                                             
        at javax.swing.JFrame.<init>(JFrame.java:189)                                                                                                                                                                                        
        at jmri.util.JmriJFrame.<init>(JmriJFrame.java:88)                                                                                                                                                                                   
        at jmri.util.JmriJFrame.<init>(JmriJFrame.java:151)                                                                                                                                                                                  
        at jmri.util.JmriJFrame.<init>(JmriJFrame.java:139)
        at jmri.jmrix.dccpp.dccppovertcp.ServerFrame.<init>(ServerFrame.java:32)
        at jmri.jmrix.dccpp.dccppovertcp.ServerFrame.getInstance(ServerFrame.java:123)
        at jmri.jmrix.dccpp.dccppovertcp.ServerAction.actionPerformed(ServerAction.java:31)
        at apps.PerformActionModel.performAction(PerformActionModel.java:31)
...
```

rhwood, 2020-07-20:
Reported working (albeit not pretty) for OP.

closed 2020-07-20 by rhwood