JMRI/JMRI #3763: Icons not displayed in web / Engine Driver if roster ID contains slash

costing, 2017-07-14:
Hi,

I was puzzled why some of the icons were not displayed for some of the entries while they were displayed ok for others. It looks like when the ID contains a slash (/) the web interface generates the request up to the first slash only and then the web server returns a 404 error for that request saying that the roster entry is not found (for a good reason, the ID is truncated...).

The workaround is easy, remove '/' from all entries, but I think this is a bug that should be addressed by correctly encoding the IDs.

Cheers,

.costin

rhwood, 2017-07-16:
Initial investigation reveals that even correctly encoded IDs get smashed into '/'s at some point outside our control (I can't tell if it is in the browser or in underlying parts of Java that we do not control). It may be that the fix is to bar '/'s from Roster entry IDs.

rhwood, 2017-07-16:
Fix from #3769 that properly encodes (and decodes) Roster Entry IDs for the web will be in 4.9.1.

@costing Thank you for reporting this.

commit 601312e8 (2017-07-16): Merge pull request #3769 from rhwood/url-encode-roster-id

URL-encode roster entry Ids in URL paths
Fixes #3763
closed 2017-07-16 by rhwood