JMRI/JMRI #3765: Can we get JMRI in Linux repositories now?

bobjacobsen, 2017-07-15:
Now that we've switched from using SerialIO and RXTX to PJC, might it be possible to get JMRI into Linux repositor(ies) sot that people can install/update it with e.g. `apt-get`?

See also #771

rhwood, 2017-07-16:
Probably not because both [Debian](https://www.debian.org/doc/packaging-manuals/java-policy/x94.html) and [Fedora](https://fedoraproject.org/wiki/Packaging:Java) appear to require that all dependencies be packaged separately or built from source with JMRI.

closed 2017-08-17 by bobjacobsen

rhwood, 2019-12-21:
Links in https://github.com/JMRI/JMRI/issues/3765#issuecomment-315595566 are outdated, so here are the updated links:

- [Debian policy for Java](https://www.debian.org/doc/packaging-manuals/java-policy/)
- [Fedora Java Packaging Guidelines](https://docs.fedoraproject.org/en-US/packaging-guidelines/Java/)