JMRI/JMRI #3776: Using "Open window" as a "Start up"  action doesn't load user preferences.

DanielBoudreau, 2017-07-16:
I've found from a user's profile configuration, that the start up action "Open Operations Trains Window" opens without the user preferences with regard to table column widths and locations. Window location and size okay.  Pretty sure other windows have the same issue if opened early.  This bug has been around for awhile, tested 4.6 and the bug was there, and is still a bug using the latest from the head.  Tested 4.4, and all is good, so this bug was introduced after 4.4.

Dan

rhwood, 2017-08-08:
I think that #3890 will fix this.

DanielBoudreau, 2017-10-25:
Not sure if you've committed your fix, but the problem still exists with the latest builds and test releases.

DanielBoudreau, 2017-10-25:
Window location and size on startup are okay, only the width of the columns isn't being restore.  Try increasing the size of any table column width, and see if the next time the window opens that it is restored correctly.  The width is correctly stored in the preference file, the problem is that the table size info isn't ready when window is opened during the startup action.  For window users a timing issue?

DanielBoudreau, 2017-11-30:
Still broke. Can you try downloading and installing JMRI on a windows PC to at least confirm that you see the problem?

rhwood, 2017-12-02:
I cannot find any code in TrainsTableFrame, OperationsFrame, or TrainsTableModel that attempts to read the stored table preferences (by calling ```JTablePersistenceManager.resetState(JTable)``` and ```JTablePersistenceManager.persist(JTable)```). Where should I be looking for these calls? If these calls are not made to bind the manager to the table, it won't happen.

DanielBoudreau, 2017-12-05:
Common code in OperationsPanel.java

    public boolean loadTableDetails(JTable table) {
        if (table.getRowSorter() == null) {
            TableRowSorter<? extends TableModel> sorter = new TableRowSorter<>(table.getModel());
            table.setRowSorter(sorter);
            // only sort on columns that are String or Integer
            for (int i = 0; i < table.getColumnCount(); i++) {
                if (table.getColumnClass(i) == String.class || table.getColumnClass(i) == Integer.class) {
                    continue; // allow sorting
                }
                sorter.setSortable(i, false);
            }
        }
        // set row height
        table.setRowHeight(new JComboBox<>().getPreferredSize().height);
        // have to shut off autoResizeMode to get horizontal scroll to work (JavaSwing p 541)
        table.setAutoResizeMode(JTable.AUTO_RESIZE_OFF);
        // give each cell a bit of space between the vertical lines and text
        table.setIntercellSpacing(new Dimension(3, 1));
        // table must have a name
        table.setName(getWindowFrameRef() + ":table"); // NOI18N
        Optional<JTablePersistenceManager> manager = InstanceManager.getOptionalDefault(JTablePersistenceManager.class);
        if (manager.isPresent()) {
            manager.get().resetState(table);
            manager.get().persist(table);
            return true;
        }
        return false;
    }

Note that everything normally works fine, including table column widths, EXCEPT if the panel is opened at startup.  Maybe the persistent manager hasn't been initiated when a panel is loaded at startup?

DanielBoudreau, 2017-12-05:
Just set a break point on my PC, and the manager is present.

DanielBoudreau, 2017-12-05:
Not sure if you or someone else has done something recently, but it seems to be working correctly with the latest build on my windows PC.

DanielBoudreau, 2017-12-05:
Downloaded and installed 4.9.8ish+jenkins+20171205T1454Z+R0d94fa6, and the column widths are restored correctly on my Windows 10 PC.  Thanks for fixing this issue.

closed 2017-12-05 by DanielBoudreau