JMRI/JMRI #3790: User Group Issue:  JMRI Crashes and burns after upgrade from 3.10.1 to 4.6

dsand47, 2017-07-18:
The root problem has been determined:  Jython 2.7.0 cannot process program files containing the UTF-8 BOM header.  When present, the **_java.lang.IllegalArgumentException: Cannot create PyString with non-byte value_** exception occurs.

This can potentially affect anyone migrating from 4.0 or earlier.

Is there a more recent version of Jython that does not have this restriction?  Is it possible to adapt the python loader to remove the BOM, if present, before invoking Jython?

 Dave Sand

dsand47, 2017-07-18:
I believe that I have found a solution using **_PushbackInputStream_** in JmriScriptEngineManager.  I still need to do some code cleanup and testing.  

Is there a better solution?

Dave Sand

bobjacobsen, 2017-07-18:
> Is it possible to adapt the python loader to remove the BOM, if present, before invoking Jython?

That still allows the possibility that the file contains real Unicode characters, in which case Jython won’t be able to handle it BOM or no BOM.

Perhaps the best we can do is to detect that the file has a BOM and put up an error dialog. That could refer them to a web page with how to fix the problem in a platform-specific way using NotePad, TextEdit, emacs/vim etc

rhwood, 2017-07-18:
There is a PR to use Jython 2.7.1 in JMRI. Have you tried it?

Is there an upstream issue against Jython 2.7.0 for this? I can't find one. I basically don't want to "fix" a script that Jython (or Python) developers have determined should be invalid, if that is the case.

dsand47, 2017-07-18:
That is a good point since UTF-8 does allow non-ascii characters.

The particular case that I am helping with involves using Microsoft Visual Studio as the IDE.  Apparently, MVS does not support UTF-8 without a BOM.  I have not heard yet whether the resulting MVS files can be plain ASCII.  The other problem is that Windows frequently defaults to UTF-8 with BOM in other applications.  

What are the ramifications of putting up an error dialog during JMRI startup during script loading triggered by Preferences >> Start Up?

Dave Sand

bobjacobsen, 2017-07-18:
> The particular case that I am helping with involves using Microsoft Visual Studio as the IDE. Apparently, MVS does not support UTF-8 without a BOM. I have not heard yet whether the resulting MVS files can be plain ASCII. 

These are a bit old, but they look like they might be helpful:

https://vlasovstudio.com/fix-file-encoding/

https://social.msdn.microsoft.com/Forums/vstudio/en-US/53ada22c-7bf0-4b62-993e-75d06bc482b8/how-save-file-as-utf8-without-signature-at-vs2015-ide?forum=vclanguage

https://stackoverflow.com/questions/5406172/utf-8-without-bom

https://stackoverflow.com/questions/5418557/how-can-i-make-visual-studio-save-all-files-as-utf-8-without-signature-on-projec

svatoun, 2019-09-15:
BTW - this may be a bug in Jython as well, there's [PEP-0263](https://www.python.org/dev/peps/pep-0263/) (final), which says:
```
...the UTF-8 signature \xef\xbb\xbf will be interpreted as 'utf-8' encoding as well (even if no magic encoding comment is given)
```

dsand47, 2019-11-04:
The BOM related error still occurs at 4.17.5 which uses Jython 2.7.1.
```
2019-11-04 09:58:37,874 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-11-04 09:58:38,993 util.Log4JUtil                        INFO  - This log is appended to file: /Users/das/Library/Preferences/JMRI/log/messages.log [main]
2019-11-04 09:58:38,994 util.Log4JUtil                        INFO  - This log is stored in file: /Users/das/Library/Preferences/JMRI/log/session.log [main]
2019-11-04 09:58:39,004 apps.Apps                             INFO  - PanelPro version 4.17.5ish+jenkins+20191030T1007Z+R8c99e77 starts under Java 1.8.0_172 on Mac OS X x86_64 v10.15.1 at Mon Nov 04 09:58:39 CST 2019 [main]
2019-11-04 09:58:41,596 apps.Apps                             INFO  - Starting with profile Randy__Sim.3edd0769 [main]
2019-11-04 09:58:41,841 node.NodeIdentity                     INFO  - Using 96f9372e-14e5-4918-9967-67a1e1cb0874 as the JMRI storage identity for profile id 3edd0769 [AWT-EventQueue-0]
2019-11-04 09:58:42,172 configurexml.ConnectionConfigXml      INFO  - CAN Simulator Started [main]
2019-11-04 09:58:42,992 apps.PerformScriptModel               INFO  - Running script /Users/das/JMRI/_Configs/Randy/scripts/startup.py [main]
2019-11-04 09:58:43,006 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_172 [main]
2019-11-04 09:58:43,014 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.1 [main]
2019-11-04 09:58:46,671 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.lang.IllegalArgumentException: Cannot create PyString with non-byte value
	at org.python.core.PyString.<init>(PyString.java:57)
	at org.python.core.PyString.<init>(PyString.java:70)
	at org.python.core.PyString.<init>(PyString.java:74)
	at org.python.core.PySyntaxError.<init>(PySyntaxError.java:27)
	at org.python.core.ParserFacade.fixParseError(ParserFacade.java:95)
	at org.python.core.ParserFacade.parseExpressionOrModule(ParserFacade.java:136)
	at org.python.util.PythonInterpreter.compile(PythonInterpreter.java:320)
	at org.python.util.PythonInterpreter.compile(PythonInterpreter.java:312)
	at org.python.jsr223.PyScriptEngine.compileScript(PyScriptEngine.java:101)
	at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:64)
	at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:249)
	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:394)
	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:331)
	at jmri.script.JmriScriptEngineManager.runScript(JmriScriptEngineManager.java:437)
	at apps.PerformScriptModel.performAction(PerformScriptModel.java:32)
	at apps.StartupActionsManager.lambda$initialize$1(StartupActionsManager.java:120)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:184)
	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:175)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1382)
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:481)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:151)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:174)
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.util.stream.ReferencePipeline.forEachOrdered(ReferencePipeline.java:423)
	at apps.StartupActionsManager.initialize(StartupActionsManager.java:118)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:392)
	at jmri.implementation.JmriConfigurationManager.lambda$load$1(JmriConfigurationManager.java:205)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1382)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:204)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:192)
	at apps.Apps.<init>(Apps.java:276)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:40)
	at apps.PanelPro.PanelPro.main(PanelPro.java:120)

```

bobjacobsen, 2020-07-25:
4.6 is seriously OBE

closed 2020-07-25 by bobjacobsen