JMRI/JMRI #3796: Normal warnings from pre-processor during Ant build

bobjacobsen, 2017-07-20:
Afte #3766, ant builds emit a series of "normal warnings":
```
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source '1.8'
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1.8'
```
(See #3766 for a bit more discussion and diagnosis)  As a low-priority cleanup task, it would be good to get rid of these.  One way to do it would be to install a customizable Ant logger c.f. https://bugzilla-attachments-210286.netbeans.org/bugzilla/attachment.cgi?id=122191

bobjacobsen, 2017-07-23:
Adding a logger can suppress the warnings, but still leaves the "2 warnings emitted" total.

bobjacobsen, 2018-01-23:
There's now a [NetBeans patch set](http://hg.netbeans.org/main-silver/rev/619fea60d4ab) that fixes this, so at some point we should update our lib//org-openide-util-lookup-RELEASE82.jar file

bobjacobsen, 2018-01-23:
But maybe that patch never reached the main path, note it's date and [this page](https://netbeans.org/bugzilla/show_bug.cgi?id=252663)

rhwood, 2018-01-23:
8.2 (RELEASE82) is the current release. 9.0 is being prepped for beta, once that is out of beta, we can evaluate moving to that release. Note that NetBeans development is moving to Apache, so the older NetBeans bug tracker may be OBE.

bobjacobsen, 2020-07-25:
Not sure when it was fixed, but the messages are no longer happening.  Thanks!

closed 2020-07-25 by bobjacobsen