JMRI/JMRI #3809: Firmware Download let JMRI 4.6 die silently

pischky, 2017-07-22:
4.6-R81496dc
Connected to LocoNet by LbServer 0.12.0.0 (oder current SVN Version of LbServer1)
CU: IntelliBox-I with 1.55 SW
Firmware download for FREDI starts but sometimes JMRI gets terminated while doing the
transfer.
There ist no message in the logfile. The last message is:
downloader.LoaderPane                 INFO  - ActionListener [AWT-EventQueue-0]

devel-bobm, 2017-07-23:
@pischky, your working environment is well beyond anything I've anticipated or tested.  There are several areas where problems could originate, and I don't have appropriate hardware/software environments with which to test.  Can you provide some JMRI LocoNet logs, with both raw data and timestamps, both for a failing case and a successful case?

pischky, 2017-07-24:
Hello Bob,
Thank you very much for your help. Here are two logfiles you request. I removed about 4 hours from the second one (I have the complete on on my disk but assume that this will not help):

[Meine_JMRI_Bahngesellschaft.zip](https://github.com/JMRI/JMRI/files/1170113/Meine_JMRI_Bahngesellschaft.zip)

I can not see any reason, or hint. May be I should switch to release 4.8 and try again. I see some changes about LocoNetOverTcp support there ?
Martin

pischky, 2017-07-24:
Last message in logfile is still:
2017-07-24 16:23:23,133 downloader.LoaderPane INFO  - ActionListener [AWT-EventQueue-0]

devel-bobm, 2017-07-25:
@pischky, the "ActionListener" log message occurs whenever any one of the radio-buttons related to the "hardware options" or the "software options" is changed.  "Touching" one of these items will change the radiobutton selection and WILL have the effect of aborting an ongoing download.  This can be shown under a LocoNet Simulator connection, using a Digitrax firmware file.  Start a download, then select an un-selected hardware option radio button.  You will see the ActionListener log message and the download will be stopped.
 
Was there any user activity on the Firmware Downloader GUI when this aborted firmware download happened?  Perhaps your method of using the tool can cause unintented/unwanted GUI interaction.
 
Current code "disables" the "select file", "download" and "verify" buttons while a "download" or a "verify" is ongoing.  Perhaps it would be wise to disable all of the GUI items, except the "abort" button, when a "download" or "verify" is ongoing.

pischky, 2017-07-25:
The "ActionListener" log message is from before start of Download: Look at the timestamps. I did not touch the GUI (AFAIK).

I just tried to press the radio buttons during download and this does not show the problem. But I also think it would be wise to disable all control but "Abort".

Anyway the main problem is that JMRI tierminates completly. The DecoderPro and all other windows are removed from the screen.

devel-bobm, 2017-07-25:
@pischky, I am stumped - I don't see how the LocoNet Firmware Download tool could cause a JMRI shutdown, especially without leaving anything else in the log.
 
It may be helpful to turn on various layers of "debugging" messages in the log file.  Perhaps some chunk of code is emitting something useful when the termination happens.  As a brute-force method, you might try making the following change to "default.lcf" in the JMRI program install directory:
 
Change
log4j.rootCategory= INFO, A1, T, R
 
to
log4j.rootCategory= DEBUG, A1, T, R
 
This will result in a VERY large log file, logging details of just about every aspect of JMRI operation.  When a fail occurs, I would think that there may be some useful information within the last 100-200 lines of the log.

pischky, 2017-07-28:
Hello, I switched to release 4.8 and did not see the problem again. But I have to admit that I did not use the firmware download that much since then (I now hat to test the release and are not creating so much new versions of FREDi I have to download).
I will watch and keep you informed.

pischky, 2017-08-01:
I think, we can close this. It never happend again in release 4.8.

closed 2017-08-01 by pischky