JMRI/JMRI #3822: Create Test Release 4.9.2

bobjacobsen, 2017-07-24:
This is the second release of the 4.9/4.10 cycle. It's intended to be released around August 12 2017 from HEAD of master.

bobjacobsen, 2017-08-12:
The release-4.9.2 branch has been created. 

From now on, please document your changes in the [jmri4.9.3.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.9.3.shtml) release note file.

Maintainers, please set the 4.9.3 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.9.2/)

geowar1, 2017-08-13:
> From now on, please document your changes in the jmri4.9.3.shtml release note file.
It doesn't look like I have write access to that suppository.

devel-bobm, 2017-08-13:
George,
 
You will not have write access to the 4.9.2 branch.  This is normal.  
 
That branch is for creation of the 4.9.2 release, and is "owned" by one of the "release creators".  It starts as a "snapshot" of JMRI/JMRI "HEAD" at some point in time.  Anything beyond that which needs to be included is "manually" selected and merged by them.  
 
If you have something that absolutely must get into the 4.2.9 release, and was not current in HEAD at the time the 4.9.2 release was created, then work with Bob J. (in this case) by e-mail or by requesting that one or more of your "more-recent" pull requests be flagged (by a repository "maintainer") as "for 4.9.2".  This will make sure that the release process is aware of the "needed" pull-request.  And the "release creator" may work directly with you to understand the particular changes and determine if they will make it into 4.9.2.
 
For the average developer, for the average "pull request" that need not be special-cased into 4.9.2, it is appropriate to continue working with HEAD in JMRI/JMRI.  Make your pull requests as you have in the past.  No changes are required on your part.

rhwood, 2017-08-13:
More to George's point, the release notes for all versions are in the master branch of JMRI/website (see https://github.com/JMRI/website/tree/master/releasenotes). Same rules for PRs apply in that repository as this one.

rhwood, 2017-08-13:
Also, if you want to make a change to a release (that hasn't been released), checkout the branch for that release, and create a PR against it, the same way PRs against master work (just a different branch). Do note that generally this is discouraged, since the release maintainer (usually Bob Jacobsen or Matt Harris) needs to perform more extensive tests and QA against any PR against those branches.

bobjacobsen, 2017-08-13:
JMRI 4.9.2 has been published

closed 2017-08-13 by bobjacobsen