JMRI/JMRI #3825: Can't create new profile

bobjacobsen, 2017-07-24:
At HEAD of master:
- Start PanelPro, having it set to allow profile selection
- Click "New" and create a new one
- Continue to start

You get the JMRI main screen, Help opens to the preferences help page, then instead of preferences opening you get a crash window:
[image]

Clicking OK closes the program.  There's no way to use that profile.  I've attached a .zip version of it.

[MERG_CBus_Simulator.zip](https://github.com/JMRI/JMRI/files/1171517/MERG_CBus_Simulator.zip)

bobjacobsen, 2017-07-24:
Console traceback for the above:
```
     [java] 2017-07-24 16:37:36,837 hrottle.AbstractWiThrottlePreferences INFO  - Could not find WiThrottle preferences file (/Users/jake/Library/Preferences/JMRI/MERG_CBus_Simulator/throttle/WiThrottlePreferences.xml).  Normal if preferences have not been saved before. [init prefs]
     [java] 2017-07-24 16:37:37,116 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [init prefs]
     [java] java.util.ServiceConfigurationError: jmri.swing.PreferencesPanel: Provider jmri.web.server.FrameServletPreferencesPanel could not be instantiated
     [java] 	at java.util.ServiceLoader.fail(ServiceLoader.java:232)
     [java] 	at java.util.ServiceLoader.access$100(ServiceLoader.java:185)
     [java] 	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:384)
     [java] 	at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
     [java] 	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
     [java] 	at apps.gui3.TabbedPreferences.init(TabbedPreferences.java:178)
     [java] 	at apps.Apps$3.lambda$run$4(Apps.java:377)
     [java] 	at apps.Apps$3$$Lambda$58/1823985821.accept(Unknown Source)
     [java] 	at java.util.Optional.ifPresent(Optional.java:159)
     [java] 	at apps.Apps$3.run(Apps.java:376)
     [java] 	at java.lang.Thread.run(Thread.java:745)
     [java] Caused by: java.lang.NullPointerException: Required nonnull default for jmri.web.server.WebServerPreferences does not exist.
     [java] 	at java.util.Objects.requireNonNull(Objects.java:228)
     [java] 	at jmri.InstanceManager.getDefault(InstanceManager.java:184)
     [java] 	at jmri.web.server.FrameServletPreferencesPanel.initComponents(FrameServletPreferencesPanel.java:71)
     [java] 	at jmri.web.server.FrameServletPreferencesPanel.<init>(FrameServletPreferencesPanel.java:57)
     [java] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
     [java] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
     [java] 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
     [java] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:408)
     [java] 	at java.lang.Class.newInstance(Class.java:438)
     [java] 	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:380)
     [java] 	... 8 more
```

bobjacobsen, 2017-07-26:
There are straight-forward ways to handle this in the `jmri.web.server.FrameServletPreferencesPanel` class by armoring against the NPE, but the code there has a comment that cautions against modifications:
```
/*============================================================================*
 * WARNING      This class contains automatically modified code.      WARNING *
 *                                                                            *
 * The method initComponents() and the variable declarations between the      *
 * "// Variables declaration - do not modify" and                             *
 * "// End of variables declaration" comments will be overwritten if modified *
 * by hand. Using the NetBeans IDE to edit this file is strongly recommended. *
 *                                                                            *
 * See http://jmri.org/help/en/html/doc/Technical/NetBeansGUIEditor.shtml for *
 * more information.                                                          *
 *============================================================================*/
```

How to handle this?

rhwood, 2017-07-26:
I’m away from the computer through Sunday (writing this on an iPhone). I can’t tell what is returning null; if someone can tell me what is returning null, I should be be able to tell what shouldn’t be null (I think that armoring the frame servlet preferences panel will probably reveal another failure for the same null object in the WebServerPreferencesPanel, so whatever is being called that returns null needs to be fixed not to.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2017-07-27:
>  I can’t tell what is returning null; if someone can tell me what is returning null, I should be be able to tell what shouldn’t be null 

If you get a chance to take a look, that would be great, but OK if not.

A InstanceManager.getDefault(WebServerPreferences.class) in jmri.web.server.FrameServletPreferencesPanel initComponents() fails because there’s no WebServerPreferences object yet and the InstanceManager doesn’t have a mechanism to provide a default one.

That’s invoked by the ServiceLoad mechanism, apparently from apps.gui3.TabbedPreferences init() invoking the ServiceLoader

rhwood, 2017-07-27:
I think that WebServerPreferences class has a getDefault() method that should be used instead of calling InstanceManager.getDefault(...). You can probably make that fix immediately, and I’ll ensure the NetBeans form editor does not stomp on it later.

Randall Wood
On my iPhone

bobjacobsen, 2017-07-27:
@rhwood Thanks!  That was the clue I needed.  WebServerPreferences's getDefault was just creating (elegantly) an object for InstanceManager, so I added the InstanceManagerAutoDefault flag interface so the InstanceManager would automatically create one when first asked for one. No changes to jmri.web.server.FrameServletPreferencesPanel needed, so no problems with changes to auto code.  PR#3835. Thanks again.

PR #3835 (2017-07-28): Fix Webserver preferences causing crash creating new Profile
In WebServerPreferences:
 - Allow InstanceManager to automatically create default object
- Deprecate local static getDefault() in favor of using default created in InstanceManager

Fixes #3825
closed 2017-07-28 by bobjacobsen