JMRI/JMRI #3846: Update for Zimo Decoders with Firmware higher than 35

alain355, 2017-07-29:
Hello Team

As discovered in Digest Number 10505, SW version (CV7=37) of the latest Zimo decoders are  higher than the current max version allowed by the Zimo_Unified_software_v32_*.xml files (highVersionID being 35). Thus JMRI can't detect the decoder family when reading CV7 and finding 37.

The current JMRI decoder list includes 5 unused Zimo_Unified_software_v34_*.xml files,  those not being shown nor used, and being replaced if used in the user's roster by the v32 definitions which includes definitions for SW version from V32 to V35.

Thus I've used those 5 V34 files to create 5 new V36 files accepting SW version from 36 to 40. I also created engines with them in current test version V4.9.1. As no updated manual was available from ZIMO, CV definition is the same than with V32 files. (Another option was to update all V32 files by allowing the SW version to go up to 40)

Can you please include them in any future version of JMRI.

AMHA the 5 Zimo_Unified_software_v34_*.xml being unused could be removed from the JMRI packages.

Regards from Paris

Alain
[ZIMO_V36.zip](https://github.com/JMRI/JMRI/files/1185094/ZIMO_V36.zip)

KenC57, 2017-07-31:
Merged via PR 3853

closed 2017-07-31 by KenC57

alain355, 2017-07-31:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; } Thanks Ken, have a nice day ;)