JMRI/JMRI #3850: Large layout startup running very slow

bobjacobsen, 2017-07-30:
With current head of master, including recent PRs, the startup time for a local layout's panel files has gone from ~15 seconds to over 10 minutes.

Thread dumps & monitoring shows a lot of time (>90%) being spent in Java2D arbitration and locking.  If I back up to JMRI 4.9.1, that's negligible.

This layout has lots of icons on a number of panels. Many of them are rotated 180 degrees, some are +/- 90 degrees. There are just two icons that are rotated by other angles, and none that are (deliberately) being scaled.

Before I spend a bunch of time with GIt bisect, does anybody have any thoughts on what might be happening here?

bobjacobsen, 2017-07-31:
Git bisect narrows the problem down to commit c4501eebab189f06261ca55496775f42eeba4e3b, see below.
```
c4501eebab189f06261ca55496775f42eeba4e3b is the first bad commit
commit c4501eebab189f06261ca55496775f42eeba4e3b
Author: George Warner <geowar1@mac.com>
Date:   Mon Jul 17 16:49:54 2017 -0700

    Turned on AntiAliasing hints

:040000 040000 1ab25906815bdd990ed86f927dca6fd8d7b13e41 1e34451f17951d35245437e3972c7935aeb52369 M	java
```

The changes are in two files:  java/src/jmri/jmrit/display/PositionableLabel.java and java/src/jmri/jmrit/display/controlPanelEditor/shape/PositionableShape.java  As the layout that's having the problem isn't using CPE, I first checked PositionableLabel.  These two lines near 1008 in the file are the problem:
```
        g2d.setRenderingHint(RenderingHints.KEY_INTERPOLATION,
               RenderingHints.VALUE_INTERPOLATION_BICUBIC);
```

Large panels (maybe 100 icons) are unusable on his machine (Windows 10 recent mid-scale) and my brand-new MacBook Pro.  Interestingly, the same lines appear at about line 805 in the file, but apparently aren't an issue there; I haven't checked what those are used for.

What problem are those two lines supposed to solve?

bobjacobsen, 2017-07-31:
The poor performance of Java `VALUE_INTERPOLATION_BICUBIC` is a known problem on Mac, see this [OpenJDK bug report](https://bugs.openjdk.java.net/browse/JDK-8017247). The situation for Windows is less clear; perhaps it's hardware specific?

rhwood, 2017-07-31:
> Git bisect narrows the problem down to commit c4501ee, see below.

IIRC, these are attempting to ensure rotated text is not jagged. Based on https://docs.oracle.com/javase/8/docs/api/java/awt/RenderingHints.html these two hints cause a most likely higher quality rendering at the cost of  speed. 

There was some discussion about caching rendering results as well. I do not recall if caching decisions changed in the same PR that contains these changes.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2017-08-06:
#3855 confirmed to fix this.

closed 2017-08-06 by bobjacobsen