JMRI/JMRI #3854: Sountraxx external include files in wrong directory

mmosher5501, 2017-07-31:
I've  noticed in 4.9.1 the main definition files of the Soundtraxx ECO and TSU2 update went into correct directory, but the 3 external include files got placed into the "tcs" directory not the "soundtraxx" directory.
Also the version note stated I did an update to TCS decoders not the correct Soundtraxx decoders.
Could not add comment to #3767 probably cause it's closed.

#3767 original issue with zip file
#3810 pull request
@bobjacobsen

bobjacobsen, 2017-08-01:
Sorry about that, didn't read the README carefully enough to notice that the "tcs" reference was probably just an oversight.  #3854 will fix this shortly. Thanks for catching it.

bobjacobsen, 2017-08-01:
I think this is fixed by #3854, now merged.  Thanks for pointing it out.

closed 2017-08-02 by bobjacobsen