JMRI/JMRI #3860: CTI/Acela NPE during config

KenC57, 2017-08-01:
When I try creating a CTI connection, I get:
---
2017-08-01 11:33:59,422 jmrix.JmrixConfigPane                 WARN  - Attempt to load jmri.jmrix.acela.serialdriver.ConnectionConfig failed: java.lang.NullPointerException: Required nonnull default for jmri.jmrix.acela.AcelaSystemConnectionMemo does not exist. [AWT-EventQueue-0]
---
This is thrown when jmrix/JmrixConfigPane line 336 catches the NPE.

From the user group...
---
I have a new Windows 10 desktop and I installed JMRI 4.8. I have successfully setup the Digitrax connection with a PR3 interface and a DCS51 command station. All works correctly. However, I still can not setup a connection to the CTI Acela interface. The dialog box does not offer a port selection at all. I then tried JMRI 4.9.1 and this does not work either. 

I verified that the Acela interface is fully functional using the CTI software, so this is not a hardware or OS issue. This was broken with the JMRI 4.* release, but worked back with JMRI 3.10. 

I complained about this some time back and another user has as well. I would appreciate that this got some attention. 

Kenneth Underwood 
---

KenC57, 2017-08-08:
One of the contributing factors is in AbstractSerialConnectionConfig: 657 it has:
`    /**
     * This is purely here for systems that do not implement the
     * SystemConnectionMemo Acela, CAN BUS, CMRI, Grapevine, QSI, Zimo
     * {@literal &} RPS and can be removed one they have been migrated
     *
     * @return Resource bundle for action model
     */
    protected ResourceBundle getActionModelResourceBundle() {
        return null;
    }`
CAN and CMRI has memos and so does the Acela. QSI is still effected. Adding the method to the acela/serialdriver/ConnectionConfig.java get it past the test at: AbstractSerialConnectionConfig line 664, but with this factory stuff, I'm still not following where the error about the nonnull memo is happening.
-ken c

KenC57, 2017-08-08:
Seems the memo isn't known to the instance manager. So with this new way of things, how/where does the instance manager learn these things??
-ken c

KenC57, 2017-08-08:
Looking back though emails I've got users saying this has been broken and not giving connection details since after 4.2.1, that was the last time CTI gave the option to pick a port to connect.
-ken c

silverailscolo, 2017-08-09:
There's a hint to the cause (and hopefully a fix) in PR #3898 which again opens the Acela Connection config and Node config cleanly.

bobjacobsen, 2017-08-09:
(From #3897) @KenC57:
> ... trying to track down a problem with the CTI/Acela connection ... Things now happen in the voodoo libraries and can't be break pointed to see what is being found or not found correctly. It used to be easy but this has made it a fruitless effort for me.

Perhaps a better approach would be to use the documentation?

The [page on system connections]((http://jmri.org/help/en/html/doc/Technical/SystemStructure.shtml) has the step-by-step sequence in it that shows who's supposed to call what when.  Start at the line that says "After it's constructed, the ConnectionConfigManager calls" (because you know that's happening because the connection starts coming it) and see where the action diverges from what's intended.

KenC57, 2017-08-10:
It looks like:
Acela NPE fix and i18n #3898
is going to fix this.

KenC57, 2017-08-11:
Saw that #3898 was merged and the daily build picked it up. Notified a couple of users with real hardware to see if any of them would give the daily a try.

KenC57, 2017-08-11:
Got message from a user testing. Seems there is a class name issue between the xml and the code.
2017-08-11 08:52:11,531 jmrix.ConnectionConfigManager         ERROR - Unable to load [Element: <connection/>] into jmri.jmrix.acela.serialdriver.configurexml.ConnectionConfigXml [main]
java.lang.ClassCastException: jmri.jmrix.acela.serialdriver.SerialDriverAdapter cannot be cast to jmri.jmrix.acela.AcelaTrafficController
	at jmri.jmrix.acela.serialdriver.configurexml.ConnectionConfigXml.unpackElement(ConnectionConfigXml.java:329)
	at jmri.jmrix.configurexml.AbstractSerialConnectionConfigXml.load(AbstractSerialConnectionConfigXml.java:111)
	at jmri.jmrix.ConnectionConfigManager.initialize(ConnectionConfigManager.java:100)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:258)
	at jmri.implementation.JmriConfigurationManager.lambda$initializeProvider$4(JmriConfigurationManager.java:254)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown Source)
	at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:253)
	at jmri.implementation.JmriConfigurationManager.lambda$initializeProvider$4(JmriConfigurationManager.java:254)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown Source)
	at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:253)
	at jmri.implementation.JmriConfigurationManager.lambda$load$1(JmriConfigurationManager.java:181)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(Unknown Source)
	at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:180)
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:168)
	at apps.Apps.<init>(Apps.java:257)
	at apps.PanelPro.PanelPro.<init>(PanelPro.java:41)
	at apps.PanelPro.PanelPro.main(PanelPro.java:104)
2017-08-11 08:52:11,537 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.jmrix.ConnectionConfigManager: Unable to create connection "CTI Electronics" (A). [main]
2017-08-11 08:52:11,901 plementation.JmriConfigurationManager ERROR - Exception initializing apps.StartupActionsManager: jmri.util.prefs.InitializationException: Unable to run startup actions due to earlier failures. [main]

bobjacobsen, 2017-08-11:
```
2017-08-11 08:52:11,531 jmrix.ConnectionConfigManager ERROR - Unable to load [Element: ] into jmri.jmrix.acela.serialdriver.configurexml.ConnectionConfigXml [main]
java.lang.ClassCastException: jmri.jmrix.acela.serialdriver.SerialDriverAdapter cannot be cast to jmri.jmrix.acela.AcelaTrafficController
```

Could you have that person "upload debugging info" please?  That'll include the contents of their configuration file.

KenC57, 2017-08-11:
I think I spotted the issue. Pushing fix.

KenC57, 2018-01-07:
I received confirmation that Acela is now working correctly. It was able to read all his nodes which included: the Acela module,  Signalman module,  Switchman module, and SY sentry module.

closed 2018-01-07 by KenC57